mavlink/MAVSDK #948: GTest is a submodule dependency, but unless all other dependencies is not automatically downloaded

LorenzMeier, 2020-01-01:
This leads to unexpected build failures on clean checkouts:

```
CMake Error at src/CMakeLists.txt:73 (add_subdirectory):
  The source directory

    /Users/lorenzmeier/src/MAVSDK/src/third_party/gtest

  does not contain a CMakeLists.txt file.
```

Running `git submodule init && git submodule update` fixes it. Having some dependencies auto-download and others not is very confusing.

JonasVautherin, 2020-01-03:
That's an interesting point. I would believe that this is one consequence of using submodules: you can never really know if you must fetch the submodules or if they are optional, can you?

The reasoning is the following:
* For some reasons we could discuss, I am not into using submodules for dependencies.
* When I removed the dependencies, I did not change gtest. Because we need submodules for `proto/` and `mavlink/`, that would not have removed the need for a `--recursive` checkout. I am not sure if moving gtest into our superbuild would make sense or not. I guess it could, because the [Google note](https://github.com/google/googletest/blob/36066cfecf79267bdf46ff82ca6c3b052f8f633c/googletest/docs/faq.md#why-is-it-not-recommended-to-install-a-pre-compiled-copy-of-google-test-for-example-into-usrlocal) advising against installing gtest may apply to other dependencies, and therefore one should probably not install dependencies coming from our superbuild.

TL;DR: we could move gtest into the superbuild, but that would not remove the need for the `--recursive` checkout.

What do you think?

LorenzMeier, 2020-01-03:
What I'm asking for is consistency: Either the super build handles all the dependencies, or it handles none and they are documented. I think removing the automation you already have would however be a mistake. It makes it convenient. If you simply add the git submodules update to the super build you're already done.

I wouldn't spend too much energy on trying to find the perfect solution. Do it one way, make it consistent and make sure the documentation is clear.

Right now my main criticism of the documentation is that it is trying to explain too much from the start (it is a reference, not a Quickstart). I don't want to read all of it. I just need to know if a) MAVSDK is for me (why should I use it) and b) give me the shell commands to get going in 3 minutes and c) give me a simple example.

That should be in the Quickstart and maybe even in the README. Look at what Homebrew does:
https://brew.sh

JonasVautherin, 2020-01-03:
While I perfectly agree with you about the documentation, I don't think I would want the superbuild to update the submodules. Sometimes I want to test with different states of the submodules (`mavlink/` and/or `proto/`).

We could add logic similar to PX4, which shows a message to the user ("you're submodules are not in the state we expect. If you're not an expert, type <key>, ..."). But this brings its own kind of complexity and confusion (which may be necessary for PX4, but which I believe is not for MAVSDK).

I would vote for merging that with #949 and I'll work on improving the documentation, if that's okay for you.

LorenzMeier, 2020-01-03:
Cool.

JonasVautherin, 2020-01-03:
Closing in favor of #949.

closed 2020-01-03 by JonasVautherin