mavlink/MAVSDK #949: Quickstart in README lies

LorenzMeier, 2020-01-01:
The Quickstart link doesn't provide a quick start, but instead points to the convoluted main page:
https://mavsdk.mavlink.io/develop/en/cpp/#getting-started

What I would expect is this:

```
git submodule init
git submodule update
mkdir -p build/default
cd build/default
cmake ../..
make
sudo -H make install
```

JonasVautherin, 2020-01-03:
Agreed. I'll spend some time on documentation ASAP :ok_hand:.

hamishwillee, 2020-01-15:
Normally a quickstart is intended for a C++ mavsdk app developer, not a contributor - right? In that case it might make sense that the instructions start from installing [release binaries](https://github.com/mavlink/MAVSDK/releases). They will often also focus on a particular platform like Linux. 

- How are these installed system wide in each case? e.g. for mavsdk_0.23.0_ubuntu16.04_amd64.deb
 is it just that you install this using the package manager and everything is set up for you going forward?
- What about windows binaries (I see the server is there, but not the main binaries).

If you can clarify I can help with this.

julianoes, 2020-01-17:
Right, we need to explain how to install it without building it.

closed 2021-03-25 by hamishwillee