mavlink/MAVSDK #952: set a GPS location to a drone that doesn't use GPS

bone3011, 2020-01-05:
I am kinda newb in MAVSDK and PX4 and every help will be appreciated.
I'd like to set a GPS location to a drone that doesn't use GPS.
mavlink has a command number 48 (SET_GPS_GLOBAL_ORIGIN).
How can I use it (if this is the right command), to send a position to the drone?
thanks

JonasVautherin, 2020-01-06:
[`SET_GPS_GLOBAL_ORIGIN`](https://mavlink.io/en/messages/common.html#SET_GPS_GLOBAL_ORIGIN) is not (yet) supported by MAVSDK, so that would need to be contributed. I personally don't know how it works exactly. I would say we first need to understand how it is supposed to be used, and then it could be contributed to MAVSDK :slightly_smiling_face:.

julianoes, 2020-01-17:
I don't think what you're describing is supported by PX4? Or is it?

If you want to use an external source as position input for PX4 you could check the [mocap plugin](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/mocap/include/plugins/mocap/mocap.h). I think it is only for local (in meter) coordinates and not global ones though.

mcsauder, 2021-01-18:
Hi everyone, these two PR's are related to this issue so I thought I'd post them here.  Once/if these are merged upstream I'll help out with work on the MAVSDK side where I can:
https://github.com/PX4/PX4-ECL/pull/962
https://github.com/PX4/PX4-Autopilot/pull/16544

bone3011, 2021-01-18:
> PX4/PX4-ECL#962 <https://github.com/PX4/PX4-ECL/pull/962>
> PX4/PX4-Autopilot#16544 <https://github.com/PX4/PX4-Autopilot/pull/16544>

thanks Mark*!*

closed 2024-05-27 by julianoes