mavlink/MAVSDK #953: Naming clash with Windows API

jehontan, 2020-01-14:
In the `ConnectionResult` enum, the value `SOCKET_ERROR` clashes with a preprocessor definition in `WinSock2.h`, causing weird substitutions and compile failure.

Propose to change the name `SOCKET_ERROR` to something else.

jehontan, 2020-01-14:
Similar issue with `ERROR` in `Mission::Result`, clashes with `ERROR` defined in `wingdi.h`.

julianoes, 2020-01-17:
Right, that's correct. However, we could usually work around it by doing `#undef`:

https://github.com/mavlink/MAVSDK/blob/164e5fe719608898b009364dbd870846d5163bfc/src/core/tcp_connection.h#L13

Does that not work for you?

jehontan, 2020-01-18:
I'm using it with another library that uses winsock, so the #undef would mess with the library.

In any case it's probably best practice to avoid clashes with the Windows API.

julianoes, 2020-01-19:
I think we can use `CamelCase` instead of `ALLCAPS` soon in the future.

JonasVautherin, 2020-01-20:
Agreed, and that would also follow [this guidelines](https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#enum5-dont-use-all_caps-for-enumerators).

I started working on auto-generating C++, and I believe that will be a good opportunity to do that :+1:.

julianoes, 2020-05-06:
Fixed after auto-generation and #1075.

closed 2020-05-06 by julianoes