mavlink/MAVSDK #959: Error: get param busy timeout: CAL_GYRO0_ID/CAL_ACC0_ID/CAL_MAG0_ID

q7frkz, 2020-01-22:
Hello,

Everytime i launch my code i have these errors :

`[03:06:12|Error] Error: get param busy timeout: CAL_GYRO0_ID (mavlink_parameters.cpp:456)
[03:06:12|Error] Error: Param for gyro cal failed. (telemetry_impl.cpp:925)
[03:06:13|Error] Error: get param busy timeout: CAL_ACC0_ID (mavlink_parameters.cpp:456)
[03:06:13|Error] Error: Param for accel cal failed. (telemetry_impl.cpp:936)
9
[03:06:13|Error] Error: get param busy timeout: CAL_MAG0_ID (mavlink_parameters.cpp:456)
[03:06:13|Error] Error: Param for mag cal failed. (telemetry_impl.cpp:947)
[03:06:14|Error] Error: get param busy timeout: SYS_HITL (mavlink_parameters.cpp:456)
[03:06:14|Error] Error: Param to determine hitl failed. (telemetry_impl.cpp:971)
`
In example there is loop which wait health pass to ok.
Mine never become ok.

I calibrate drone using Mission planer, no problem.
I calibrate drone using QGC, no problem.
Impossible to calibrate with example, first stage, program asks to tape any key but doesnt detect any key:

`$:~/Desktop/MAVSDK/example/calibrate/build$ sudo ./calibrate serial:///dev/ttyTHS0:1500000
[04:13:39|Info ] MAVSDK version: 0.23.0-21-gd9ecfcb6 (mavsdk_impl.cpp:25)
Waiting to discover system...
[04:13:39|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:401)
[04:13:39|Debug] Component Autopilot (1) added. (system_impl.cpp:386)
[04:13:39|Debug] Discovered 1 component(s) (UUID: 1) (system_impl.cpp:556)
Discovered system with UUID: 1
Calibrating accelerometer...
[04:13:40|Debug] MAVLink: info: Initialising APM (system_impl.cpp:292)
[04:13:40|Warn ] sending again, retries to do: 3  (241). (mavlink_commands.cpp:222)
[04:13:40|Debug] MAVLink: critical: Place vehicle level and press any key. (system_impl.cpp:292)
\n      //try enter
   \n   //try space
dfsdf\n  //try other
hdhgf\n
hgsgdf\n
^C
$:~/Desktop/MAVSDK/example/calibrate/build$
`

I don't know what to do.

Best regards

julianoes, 2020-01-23:
It looks like you are using ArduPilot and not PX4. ArduPilot is currently not supported by MAVSDK, also see #728.

closed 2020-01-23 by julianoes

q7frkz, 2020-01-28:
Hello

it seems ardupilot is installed yes but i dont find how to install px4

can you help me ?
With mission planer, installation ok but crash 5-6 seconds after boot
With QGC impossible to update, pixhawk only detected with bugged px4 installed via mission planner