mavlink/MAVSDK #977: unit_tests_backend stall

julianoes, 2020-01-29:
The backend unit test stalled mysteriously:

```
[----------] 36 tests from CameraServiceImplTest
[ RUN      ] CameraServiceImplTest.takesPhotoEvenWhenArgsAreNull
```

```
(gdb) thread apply all bt

Thread 3 (Thread 0x7f7f070c7700 (LWP 100460)):
#0  0x00007f7f0805fd45 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x00007f7f08217f1d in gpr_cv_wait (cv=0x27dc978, mu=0x27dc940, abs_deadline=...) at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/gpr/sync_posix.cc:128
#2  0x00007f7f08b53591 in grpc_core::Executor::ThreadMain (arg=0x27dc940) at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/iomgr/executor.cc:231
#3  0x00007f7f0821cdc6 in grpc_core::(anonymous namespace)::ThreadInternalsPosix::ThreadInternalsPosix(char const*, void (*)(void*), void*, bool*, grpc_core::Thread::Options const&)::{lambda(void*)#1}::operator()(void*) const (this=0x27c7f40, v=0x27c7f40) at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/gprpp/thd_posix.cc:140
#4  0x00007f7f0821cc85 in grpc_core::(anonymous namespace)::ThreadInternalsPosix::ThreadInternalsPosix(char const*, void (*)(void*), void*, bool*, grpc_core::Thread::Options const&)::{lambda(void*)#1}::__invoke(void*) (v=0x27c7f40) at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/gprpp/thd_posix.cc:110
#5  0x00007f7f080594e2 in start_thread () from /lib64/libpthread.so.0
#6  0x00007f7f07a5f6d3 in clone () from /lib64/libc.so.6

Thread 2 (Thread 0x7f7f078c8700 (LWP 100459)):
#0  0x00007f7f0805fd45 in pthread_cond_wait@@GLIBC_2.3.2 () from /lib64/libpthread.so.0
#1  0x00007f7f08217f1d in gpr_cv_wait (cv=0x27dbd78, mu=0x27dbd40, abs_deadline=...) at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/gpr/sync_posix.cc:128
#2  0x00007f7f08b53591 in grpc_core::Executor::ThreadMain (arg=0x27dbd40) at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/iomgr/executor.cc:231
#3  0x00007f7f0821cdc6 in grpc_core::(anonymous namespace)::ThreadInternalsPosix::ThreadInternalsPosix(char const*, void (*)(void*), void*, bool*, grpc_core::Thread::Options const&)::{lambda(void*)#1}::operator()(void*) const (this=0x27b62a0, v=0x27b62a0) at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/gprpp/thd_posix.cc:140
#4  0x00007f7f0821cc85 in grpc_core::(anonymous namespace)::ThreadInternalsPosix::ThreadInternalsPosix(char const*, void (*)(void*), void*, bool*, grpc_core::Thread::Options const&)::{lambda(void*)#1}::__invoke(void*) (v=0x27b62a0) at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/gprpp/thd_posix.cc:110
#5  0x00007f7f080594e2 in start_thread () from /lib64/libpthread.so.0
#6  0x00007f7f07a5f6d3 in clone () from /lib64/libc.so.6

Thread 1 (Thread 0x7f7f078c9840 (LWP 100458)):
#0  0x00007f7f08ba447a in grpc_core::TraceFlagList::Set (name=0x27df1b0 "", enabled=true) at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/debug/trace.cc:58
#1  0x00007f7f08ba4742 in parse (s=0x27df190 "") at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/debug/trace.cc:130
#2  0x00007f7f08ba4679 in grpc_tracer_init () at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/debug/trace.cc:147
#3  0x00007f7f084464e5 in grpc_init () at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/core/lib/surface/init.cc:159
#4  0x00007f7f08a614e1 in grpc::internal::GrpcLibrary::init (this=0x2762900) at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/include/grpcpp/impl/grpc_library.h:34
#5  0x00007f7f0989789b in grpc::GrpcLibraryCodegen::GrpcLibraryCodegen (this=0x27abf50, call_grpc_init=true) at /home/julianoes/src/MAVSDK/build/third_party/install/include/grpcpp/impl/codegen/grpc_library.h:45
#6  0x00007f7f09897138 in grpc_impl::CompletionQueue::CompletionQueue (this=0x27abf50, attributes=...) at /home/julianoes/src/MAVSDK/build/third_party/install/include/grpcpp/impl/codegen/completion_queue_impl.h:244
#7  0x00007f7f08aab268 in grpc_impl::ServerCompletionQueue::ServerCompletionQueue (this=0x27abf50, completion_type=GRPC_CQ_NEXT, polling_type=GRPC_CQ_DEFAULT_POLLING, shutdown_cb=0x0) at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/include/grpcpp/impl/codegen/completion_queue_impl.h:407
#8  0x00007f7f08aa9cb0 in grpc_impl::ServerBuilder::BuildAndStart (this=0x7ffd003c7130) at /home/julianoes/src/MAVSDK/build/third_party/grpc/grpc/src/grpc/src/cpp/server/server_builder.cc:298
#9  0x00000000007a304c in (anonymous namespace)::CameraServiceImplTest::SetUp (this=0x27c7f70) at /home/julianoes/src/MAVSDK/src/backend/test/camera_service_impl_test.cpp:59
#10 0x00007f7f07df36e4 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void> (object=0x27c7f70, method=&virtual testing::Test::SetUp(), location=0x7f7f07dfdf91 "SetUp()") at /home/julianoes/src/MAVSDK/src/third_party/gtest/googletest/src/gtest.cc:2593
#11 0x00007f7f07dd90ab in testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void> (object=0x27c7f70, method=&virtual testing::Test::SetUp(), location=0x7f7f07dfdf91 "SetUp()") at /home/julianoes/src/MAVSDK/src/third_party/gtest/googletest/src/gtest.cc:2629
#12 0x00007f7f07db857c in testing::Test::Run (this=0x27c7f70) at /home/julianoes/src/MAVSDK/src/third_party/gtest/googletest/src/gtest.cc:2663
#13 0x00007f7f07db932d in testing::TestInfo::Run (this=0x277c790) at /home/julianoes/src/MAVSDK/src/third_party/gtest/googletest/src/gtest.cc:2845
#14 0x00007f7f07db9a6c in testing::TestSuite::Run (this=0x276ae90) at /home/julianoes/src/MAVSDK/src/third_party/gtest/googletest/src/gtest.cc:2977
#15 0x00007f7f07dc650f in testing::internal::UnitTestImpl::RunAllTests (this=0x2775730) at /home/julianoes/src/MAVSDK/src/third_party/gtest/googletest/src/gtest.cc:5518
#16 0x00007f7f07df6604 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (object=0x2775730, method=(bool (testing::internal::UnitTestImpl::*)(testing::internal::UnitTestImpl * const)) 0x7f7f07dc60b0 <testing::internal::UnitTestImpl::RunAllTests()>, location=0x7f7f07dfe819 "auxiliary test code (environments or event listeners)") at /home/julianoes/src/MAVSDK/src/third_party/gtest/googletest/src/gtest.cc:2593
#17 0x00007f7f07ddb94b in testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (object=0x2775730, method=(bool (testing::internal::UnitTestImpl::*)(testing::internal::UnitTestImpl * const)) 0x7f7f07dc60b0 <testing::internal::UnitTestImpl::RunAllTests()>, location=0x7f7f07dfe819 "auxiliary test code (environments or event listeners)") at /home/julianoes/src/MAVSDK/src/third_party/gtest/googletest/src/gtest.cc:2629
#18 0x00007f7f07dc6066 in testing::UnitTest::Run (this=0x7f7f07e28198 <testing::UnitTest::GetInstance()::instance>) at /home/julianoes/src/MAVSDK/src/third_party/gtest/googletest/src/gtest.cc:5105
#19 0x00000000007a07c1 in RUN_ALL_TESTS () at /home/julianoes/src/MAVSDK/src/third_party/gtest/googletest/include/gtest/gtest.h:2472
#20 0x00000000007a07a8 in main (argc=1, argv=0x7ffd003c78d8) at /home/julianoes/src/MAVSDK/src/backend/test/backend_main.cpp:6
```

julianoes, 2020-01-29:
This looks somewhat like:
https://github.com/grpc/grpc/issues/21213
and might be fixed with:
https://github.com/grpc/grpc/pull/21662

julianoes, 2020-01-29:
This happens to me on Fedora 31 when compiling with clang.

julianoes, 2020-01-29:
The issue is resolved when switching to from v1.26.0 to [v1.27.0-pre2](https://github.com/grpc/grpc/releases/tag/v1.27.0-pre2).

julianoes, 2020-01-29:
It also happens reproducible with GCC. How was this not caught by CI?

julianoes, 2020-01-30:
Still happening, until we update gRPC.

veblush, 2020-02-07:
Hey Julian, did you open this because it's still happening with v1.27?

julianoes, 2020-02-07:
I re-opened this because we did not actually update to v1.27. All #978 had done was just adding a unit test back to CI. And I hoped it would cause CI to fail until we update but the stall somehow did not happen in CI.

Anyway, I think we can close this once that we update to v1.27. I was just waiting on the release which I see has now been made: https://github.com/grpc/grpc/releases. Let me update! :smile:

PR #978 (2020-01-30): Fix gRPC stall
In this pull request I want to fix #977. First step is to enable the backend test to see if it reproduces as well. Then, we can hopefully update the gRPC version to fix it.
PR #982 (2020-05-07): third_party: update gRPC
This should fix a stall that we have seen in the backend unit test.

Now really fixes #977.
closed 2020-05-07 by julianoes