mavlink/MAVSDK #980: mavshell example is not starting shell

shrit, 2020-01-31:
Hello,

I have just compiled mavshell example. After discovering the system I am getting this error in the real px4 shell:
`pxh> ERROR [mavlink] Failed to start shell (-1)`

On the mavshell side only  a blocking terminal
PX4 version 1.9.2
Any ideas??

shrit, 2020-01-31:
I suppose that this shell does not work in SITL simulation as it requires Nuttx to be used instead of Linux

JonasVautherin, 2020-02-01:
Exact, it does not run on SITL. Did you try on NuttX?

shrit, 2020-02-01:
I no longer have access to Pixhawk hardware.
Unless if there is a magic virtual machine for running SITL with PX4 inside Nuttx

JonasVautherin, 2020-02-01:
Right. Well, closing that, as I know for a fact that mavshell only runs in NuttX :slightly_smiling_face:. Feel free to reopen if necessary!

closed 2020-02-01 by JonasVautherin