mavlink/MAVSDK #981: Missing stropts.h is stopping the installation of  cURL

shrit, 2020-02-01:
Hello,

Since glibc version 2.30, `stropts.h` has been removed. At the same time, this curl version should not require stropts.h to exist as described in the source code https://github.com/curl/curl/blob/28b5b1c20a7f38b079b8828d06423d4b7a3ed439/lib/if2ip.c#L46. 
```
/meta/MAVSDK/build/default/third_party/curl/curl/src/curl/lib/if2ip.c:47:12: fatal error: stropts.h: No such file or directory
   47 | #  include <stropts.h>
      |            ^~~~~~~~~~~
```
This is blocking the entire installation of MAVSDK on Arch Linux.

JonasVautherin, 2020-02-02:
I'm on Arch Linux, I have glibc 2.30, and curl builds without an issue for me :thinking:. Do you have an idea how I could reproduce?

shrit, 2020-02-02:
I do not know what I am missing I never had a problem installing the SDK
I am installing the SDK directly from the master version.
My arch version is `5.4.2-arch1-1`
You can search for `<stropts.h>` using `locate` or `find` to see if it is still on your system.
The installation command I am using is: 
`cmake -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=ON -Bbuild/default -H.`

The output of the command:
```
develop /meta/MAVSDK> cmake -DCMAKE_BUILD_TYPE=Debug -DBUILD_SHARED_LIBS=ON -Bbuild/default -H.
-- Preparing external project "jsoncpp" with args:
-- -- -DBUILD_SHARED_LIBS=ON
-- -- -DCMAKE_BUILD_TYPE=Debug
-- -- -DPC_JSONCPP_CFLAGS=-I/meta/MAVSDK/build/default/third_party/install/include
-- -- -DPC_JSONCPP_CFLAGS_I=
-- -- -DPC_JSONCPP_CFLAGS_OTHER=
-- -- -DPC_JSONCPP_FOUND=1
-- -- -DPC_JSONCPP_INCLUDEDIR=/meta/MAVSDK/build/default/third_party/install/include
-- -- -DPC_JSONCPP_INCLUDE_DIRS=/meta/MAVSDK/build/default/third_party/install/include
-- -- -DPC_JSONCPP_LDFLAGS=-L/meta/MAVSDK/build/default/third_party/install/lib64
-- -- -ljsoncpp
-- -- -DPC_JSONCPP_LDFLAGS_OTHER=
-- -- -DPC_JSONCPP_LIBDIR=/meta/MAVSDK/build/default/third_party/install/lib64
-- -- -DPC_JSONCPP_LIBRARIES=jsoncpp
-- -- -DPC_JSONCPP_LIBRARY_DIRS=/meta/MAVSDK/build/default/third_party/install/lib64
-- -- -DPC_JSONCPP_LIBS=
-- -- -DPC_JSONCPP_LIBS_L=
-- -- -DPC_JSONCPP_LIBS_OTHER=
-- -- -DPC_JSONCPP_LIBS_PATHS=
-- -- -DPC_JSONCPP_PREFIX=
-- -- -DPC_JSONCPP_STATIC_CFLAGS=-I/meta/MAVSDK/build/default/third_party/install/include
-- -- -DPC_JSONCPP_STATIC_CFLAGS_I=
-- -- -DPC_JSONCPP_STATIC_CFLAGS_OTHER=
-- -- -DPC_JSONCPP_STATIC_INCLUDE_DIRS=/meta/MAVSDK/build/default/third_party/install/include
-- -- -DPC_JSONCPP_STATIC_LDFLAGS=-L/meta/MAVSDK/build/default/third_party/install/lib64
-- -- -ljsoncpp
-- -- -DPC_JSONCPP_STATIC_LDFLAGS_OTHER=
-- -- -DPC_JSONCPP_STATIC_LIBDIR=
-- -- -DPC_JSONCPP_STATIC_LIBRARIES=jsoncpp
-- -- -DPC_JSONCPP_STATIC_LIBRARY_DIRS=/meta/MAVSDK/build/default/third_party/install/lib64
-- -- -DPC_JSONCPP_STATIC_LIBS=
-- -- -DPC_JSONCPP_STATIC_LIBS_L=
-- -- -DPC_JSONCPP_STATIC_LIBS_OTHER=
-- -- -DPC_JSONCPP_STATIC_LIBS_PATHS=
-- -- -DPC_JSONCPP_VERSION=1.8.4
-- -- -DPC_JSONCPP_jsoncpp_INCLUDEDIR=
-- -- -DPC_JSONCPP_jsoncpp_LIBDIR=
-- -- -DPC_JSONCPP_jsoncpp_PREFIX=
-- -- -DPC_JSONCPP_jsoncpp_VERSION=
-- -- -D__pkg_config_arguments_PC_JSONCPP=jsoncpp
-- -- -D__pkg_config_checked_PC_JSONCPP=1
-- -- -Dgenerated_dir=/meta/MAVSDK/build/default/src/third_party/gtest/googlemock/gtest/generated
-- -- -Dprefix_result=/meta/MAVSDK/build/default/third_party/install/lib64
-- -- -Dtargets_export_name=GTestTargets
-- -- -DCMAKE_INSTALL_PREFIX:PATH=/meta/MAVSDK/build/default/third_party/install
-- -- -DCMAKE_POSITION_INDEPENDENT_CODE=ON
-- -- -DBUILD_SHARED_LIBS=OFF
-- -- -DJSONCPP_WITH_CMAKE_PACKAGE=ON
-- -- -DJSONCPP_WITH_TESTS=OFF
-- Configuring done
-- Generating done
-- Build files have been written to: /meta/MAVSDK/build/default/third_party/jsoncpp
[ 12%] Performing update step for 'jsoncpp'
[ 25%] Performing configure step for 'jsoncpp'
-- JsonCpp Version: 1.8.4
-- Configuring done
-- Generating done
-- Build files have been written to: /meta/MAVSDK/build/default/third_party/jsoncpp/jsoncpp/src/jsoncpp-build
[ 37%] Performing build step for 'jsoncpp'
[100%] Built target jsoncpp
[... 223 lines cut ...]
[  5%] Building C object lib/CMakeFiles/libcurl.dir/curl_rtmp.c.o
[  5%] Building C object lib/CMakeFiles/libcurl.dir/curl_gethostname.c.o
[  5%] Building C object lib/CMakeFiles/libcurl.dir/gopher.c.o
[  5%] Building C object lib/CMakeFiles/libcurl.dir/idn_win32.c.o
/meta/MAVSDK/build/default/third_party/curl/curl/src/curl/lib/if2ip.c:47:12: fatal error: stropts.h: No such file or directory
   47 | #  include <stropts.h>
      |            ^~~~~~~~~~~
compilation terminated.
make[5]: *** [lib/CMakeFiles/libcurl.dir/build.make:219: lib/CMakeFiles/libcurl.dir/if2ip.c.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[4]: *** [CMakeFiles/Makefile2:359: lib/CMakeFiles/libcurl.dir/all] Error 2
make[3]: *** [Makefile:130: all] Error 2
make[2]: *** [CMakeFiles/curl.dir/build.make:115: curl/src/curl-stamp/curl-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/curl.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
CMake Error at third_party/cmake/build_target.cmake:48 (message):
  /meta/MAVSDK/build/default/third_party/curl failed to build!
Call Stack (most recent call first):
  third_party/CMakeLists.txt:17 (build_target)

-- Configuring incomplete, errors occurred!
See also "/meta/MAVSDK/build/default/CMakeFiles/CMakeOutput.log".
See also "/meta/MAVSDK/build/default/CMakeFiles/CMakeError.log".
exit 1
```

shrit, 2020-02-02:
I was able to configure and compile the last master version of cURL downloaded directly from github

shrit, 2020-02-02:
I have also changed the downloaded version in the CMakeLists.txt the last one, and to the git master, I got the same error for both of them when configuring te SDK

julianoes, 2020-02-03:
It builds fine for me, and I'm on latest Arch Linux and I do not have `stropts.h` on my system either.

The place where it fails for you it actually has a define if `stropts.h` is available:

https://github.com/curl/curl/blob/curl-7_67_0/lib/if2ip.c#L46-L48

and the configure step correctly determines that for me:
```
-- Looking for 41 include files stdio.h, ..., stropts.h
-- Looking for 41 include files stdio.h, ..., stropts.h - not found
```

Have you tried wiping the `build` folder and trying fresh?

shrit, 2020-02-03:
Thank you, it is working now.
That is was obvious and tricky at the same time.

closed 2020-02-03 by julianoes