mavlink/MAVSDK #992: takeoff_and_land error (fatal error: plugin_base.h: No such file or directory)

Gardiar, 2020-02-18:
I coppied and pasted example code from "https://mavsdk.mavlink.io/develop/en/examples/takeoff_and_land.html" to my "~/catkin_ws/src/droneControl/localization.cpp" folder. When I build catkin_ws with "catkin build" I take error like that:

**`

- Errors     << droneControl:make /home/yusufozben/catkin_ws/logs/droneControl/build.make.025.log          
- In file included from /home/yusufozben/catkin_ws/src/droneControl/localization.cpp:9:0:
- /usr/include/mavsdk/plugins/action/action.h:6:10: fatal error: plugin_base.h: No such file or directory
-  #include "plugin_base.h"
-           ^~~~~~~~~~~~~~~
- compilation terminated.
- make[2]: *** [CMakeFiles/localization.dir/localization.cpp.o] Error 1
- make[1]: *** [CMakeFiles/localization.dir/all] Error 2
- make: *** [all] Error 2
- cd /home/yusufozben/catkin_ws/build/droneControl; catkin build --get-env droneControl | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

`**

mavsdk location : "/usr/include/mavsdk"

I try to add plugin_base.h directory to cmakefile but it doesn't run. I also try to do this "https://github.com/mavlink/MAVSDK/issues/802".

JonasVautherin, 2020-02-18:
Did you solve it? :slightly_smiling_face:

Gardiar, 2020-02-19:
Thanks for your reply. I solved it. 🙂

julianoes, 2020-02-19:
@Gardiar just for someone seeing this later: how did you fix it?

Gardiar, 2020-02-20:
It was an only typo in 'CMakeLists.txt'. After coping and pasting 'CMakeFiles' from 'https://github.com/mavlink/MAVSDK/blob/develop/example/takeoff_land/CMakeLists.txt', the error was fixed.

maximusron, 2021-08-08:
> It was an only typo in 'CMakeLists.txt'. After coping and pasting 'CMakeFiles.txt' from 'https://github.com/mavlink/MAVSDK/blob/develop/example/takeoff_land/CMakeLists.txt', the error was fixed.

Can you please share the fix, the github link doesnt exist anymore

JonasVautherin, 2021-08-09:
You'll find plenty of CMakeLists in the examples folder: https://github.com/mavlink/MAVSDK/tree/main/examples

AshwinDisa, 2022-01-24:
I'm facing the same exact issue, the above solutions does not solve it. Please help!

In file included from /home/ashd/MAVSDK/examples/takeoff_and_land/takeoff_and_land.cpp:8:
/usr/include/mavsdk/plugins/action/action.h:16:10: fatal error: plugin_base.h: No such file or directory
 #include "plugin_base.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/takeoff_and_land.dir/build.make:62: recipe for target 'CMakeFiles/takeoff_and_land.dir/takeoff_and_land.cpp.o' failed
make[2]: *** [CMakeFiles/takeoff_and_land.dir/takeoff_and_land.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/takeoff_and_land.dir/all' failed
make[1]: *** [CMakeFiles/takeoff_and_land.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

JonasVautherin, 2022-01-24:
Can you please open a new issue, describe which version you are using and what you did to get this error message?

maximusron, 2022-01-24:
CMake is probably not including it right, I will mail over a sample Cmake file that includes mavsdk correctly

maximusron, 2022-01-24:
Try adding   -I/usr/include/mavsdk to your compile options in the CMake file, and

        MAVSDK::mavsdk
        MAVSDK::mavsdk_action
        MAVSDK::mavsdk_telemetry
        MAVSDK::mavsdk_mission

to your target link libraries.

AshwinDisa, 2022-01-24:
This solved my issue. Thanks @maximusron

skaantosun, 2022-02-01:
Hello, we have same problem but we did not understand how to solve it. What does -I/usr/include/mavsdk mean?

JonasVautherin, 2022-02-01:
It actually may be a bug, that we could solve if somebody opened a new issue (this one is from 2020, so that's probably not exactly the same problem), explained the steps to reproduce it, and showed the error message. Just saying 😇.

maximusron, 2022-02-02:
-I/usr/include/mavsdk is the include directory of your mavsdk install that contains various headers. It is probably required for most mavsdk related CMake compilations without which objects like MAVSDK::action wouldn't be compiled. Adding it to your compile options allows it to complile with mavsdk files included like action.h, mission.h, telemetry.h, plugin_base.h to name a few

maximusron, 2022-02-02:
They're just incorrect ways of configuring a CMake file, although there should be a way that a mavsdk install doesn't require you to configure CMake like that.
But since it's a library I wouldn't expect that to be so.

julianoes, 2022-02-02:
@maximusron I'm lacking context here. Please describe the problem using a minimum example with some steps on how to reproduce it. Thank you.

julianoes, 2022-03-06:
> objects like MAVSDK::action wouldn't be compiled.

MAVSDK::action is no longer a thing, please check [these changes](https://mavsdk.mavlink.io/main/en/cpp/api_changes.html#10) and update to the latest version.

closed 2022-03-06 by julianoes

tank104, 2022-09-21:
To anyone else that has issues with this.  The problem is the main branch is later than the packages.
So the Quick Start example should be something like this:

--Get latest published version (at this time)
wget https://github.com/mavlink/MAVSDK/releases/download/v1.4.6/libmavsdk-dev_1.4.6_ubuntu20.04_amd64.deb
sudo dpkg -i libmavsdk-dev_1.4.6_ubuntu20.04_amd64.deb

--Get repo and set to tag matching release
git clone https://github.com/mavlink/MAVSDK.git --recursive
cd MAVSDK
git checkout tags/v1.4.3 -b 1.4

--then examples should work
cd examples/takeoff_and_land/
cmake -Bbuild -H.
cmake --build build -j4