mavlink/MAVSDK #994: Subscribing to mavlink of other vehicles from the same udp port

Jaeyoung-Lim, 2020-02-22:
I am trying to use other vehicle's position / velocity information to de-conflict paths when operating multiple vehicles. However, mavsdk seems to assume that there can be only one system on a single udp port.

https://github.com/mavlink/MAVSDK/blob/e7f80158d7fc4f9526d3641eb595440777005dc6/src/core/mavsdk_impl.cpp#L291-L312

This means that in order to subscribe to other vehicles (vehicles with different sysIDs) we need to have prior knowledge of what ports the vehicle will be located to. However, such circumstance is unrealistic and I believe most realistic scenarios it would be even hard to know how many vehicles would be in the network. 

Would there be a way to subscribe mavlnk information of multiple vehicles from a single udp port? or if we this need to be added, how could it be done?

hamishwillee, 2020-02-23:
That should be perfectly possible - mavlink differentiates vehicles my SysId/component id, not UDP address. Whether or not it handles the case is another matter.

JonasVautherin, 2020-02-23:
You are quoting `get_system()`, but I think you are interested in [`get_system(const uint64_t uuid)`](https://github.com/mavlink/MAVSDK/blob/e7f80158d7fc4f9526d3641eb595440777005dc6/src/core/mavsdk_impl.cpp#L314-L324), which allows you to get the system that corresponds to a UUID. 

And you get a list of discovered systems with [`register_on_discover`](https://github.com/mavlink/MAVSDK/blob/e7f80158d7fc4f9526d3641eb595440777005dc6/src/core/mavsdk.h#L230).

Does that help?

Jaeyoung-Lim, 2020-02-26:
@JonasVautherin I think that helps. I will give it another go. Thanks!

julianoes, 2020-11-20:
The API to access multiple vehicles has been changed. I think this issue has been resolved, otherwise please re-open @Jaeyoung-Lim.

closed 2020-11-20 by julianoes