mavlink/MAVSDK #996: Promise already satisfied

Katawann, 2020-02-24:
I installed MAVSDK using the debian package of the last release (mavsdk_0.24.0_ubuntu16.04_amd64.deb)

I would like to work with promise/future to manage my applicaiton but I got an error when I try using it. I took a simple example of [takeoff_and_land](https://mavsdk.mavlink.io/develop/en/examples/takeoff_and_land.html)

and I changed this line: 

```
    // Check if vehicle is ready to arm
    while (telemetry->health_all_ok() != true) {
        std::cout << "Vehicle is getting ready to arm" << std::endl;
        sleep_for(seconds(1));
    }

```
by this line (proposed in the [doc](https://mavsdk.mavlink.io/develop/en/guide/telemetry.html)):

```
{
    std::cout << "Waiting for system to be ready" << std::endl;
    auto prom = std::make_shared<std::promise<void>>();
    auto future_result = prom->get_future();
    telemetry->health_all_ok_async(
    [prom](bool result) {
        //fulfill promise if health is OK
        if (result) {// health OK
            prom->set_value();
        }
    });

    future_result.get(); //Block until promise is fulfilled.
    // Now ready to arm
}
```

but I finished with this error:

```
terminate called after throwing an instance of 'std::future_error'
  what():  std::future_error: Promise already satisfied
Aborted (core dumped)
```

Am I doing something wrong or this is an known/unknown error ?

JonasVautherin, 2020-02-25:
"Promise already satisfied" says that `prom->set_value();` is called more than once. And this happens the second time your callback (the lambda you give to `health_all_ok_async(...)`) is called (because this callback will be called multiple times, possibly with the updated "health" status).

So you need to make sure that `prom->set_value();` is called only the first time, and possibly unregister the health callback if you don't need it after that :+1:.

Katawann, 2020-02-25:
@JonasVautherin  It's working when I unregistered the callback with:
`telemetry->health_all_ok_async(nullptr);`

@hamishwillee  Not sure if it is the "prettiest" solution but it will be nice to add a note or something in the doc to say we should unregister to avoid to call the promise twice...

hamishwillee, 2020-02-25:
Hi @Katawann 
Sounds good to me. Probably should make sure that the documentation and example are updated with a comment.

Can you add a section that you think would have solved this for you? (PR easy to create by clicking button at top of page). 

[image]

Perhaps in here https://mavsdk.mavlink.io/develop/en/guide/telemetry.html#blocking-telemetry-requests-state-management

Katawann, 2020-02-25:
Sure!

I close this issue and update the doc according to this

closed 2020-02-25 by Katawann