mavlink/MAVSDK #997: Python wheel for RPI

GoMartin300, 2020-02-28:
I am currently trying to run MAVSDK within Docker containers on RPi 4.
I have it working outside of Docker by building it using the guidelines in this post:
https://discuss.px4.io/t/talking-to-a-px4-fmu-with-a-rpi-via-serial-nousb-ardunope/14119
I am doing this because pip install mavsdk is not supported for the RPi at this time.

The post does mention that there is some interest in creating a python wheel for RPi.
As this does not yet exist I am currently looking into creating a base Docker image that contains an already built version of MavSDK-python and python3. Hopefully this way I don't need to built mavSDK every time I want to build a Docker Image with MAVsdk included.

Are there any plans for supporting/enabling pip install mavsdk on RPi?
And even if supported/enabled, would it no longer require the long built times?

Until then, when I get my base Image for python 3 + mavSDK-python working, I will comment my findings in this issue.

(This is my first GitHub issue ever, please let me know if I am not following styling guidelines, formatting or other conventions,...)

julianoes, 2020-02-29:
> pip install mavsdk on RPi?

The problem is that Linux builds on PyPi are built using `manylinux1` or `manylinux2010`. They are `x86_64` based and therefore don't include `arm` binaries. This might change with `manylinux2014`, check https://github.com/mavlink/MAVSDK-Python/issues/117.

What we could/should do alternatively, is provide a RPi .deb package to install the `mavsdk_server` binary which is all that is needed for the RPi platform. You could then use that within docker or directly.

GoMartin300, 2020-03-01:
I don't have any experience with these .deb packages.
I did manage to create a Docker container based on a standard Python container and install mavsdk into it. I've placed this as an image on docker hub under the name mavsdk-python-armv8 in a private repository as well. I don't want to confuse people into using it for any kind of critical application, that's why its private for now. I could make it public with a clear description, just let me know.

I am however convinced that the .deb solution would be better, my current Docker image is 3GB in uncompressed size...
Would it be feasible for  me to create this with some research (again, no experience with .deb)? Or is this coming in the near future anyway?

For now i will leave my Dockerfile here:
```
FROM python:3

# Set the working directory, this is the image file system
WORKDIR /usr/src/app

RUN apt update
RUN apt upgrade

RUN apt-get update -y
RUN apt-get install cmake build-essential colordiff git doxygen -y 

RUN git clone https://github.com/mavlink/MAVSDK.git
WORKDIR /usr/src/app/MAVSDK
RUN git checkout master
RUN git submodule update --init --recursive
RUN cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -Bbuild/default -H.
RUN cmake --build build/default
RUN cmake --build build/default --target install
RUN ldconfig

# Building the mavsdk-server
RUN cmake -DBUILD_BACKEND=ON --symlink-install --cmake-args "-DCMAKE_SHARED_LINKER_FLAGS='-latomic'" "-DCMAKE_EXE_LINKER_FLAGS='-latomic'" -Bbuild/default -H.
RUN cmake --build build/default

WORKDIR /usr/src/app/
RUN git clone https://github.com/mavlink/MAVSDK-Python --recursive
WORKDIR /usr/src/app/MAVSDK-Python/proto/pb_plugins
RUN pip3 install --no-cache-dir -r requirements.txt
# Needs to be the -e (editable) version or doesn't work, don't know why though
RUN pip3 install --no-cache-dir -e .

WORKDIR /usr/src/app/MAVSDK-Python
RUN pip3 install --no-cache-dir -r requirements.txt -r requirements-dev.txt
RUN ./other/tools/run_protoc.sh
# Needs to be the -e (editable) version or doesn't work, don't know why though
RUN pip3 install --no-cache-dir -e .
RUN mv /usr/src/app/MAVSDK/build/default/src/backend/src/mavsdk_server /usr/src/app/MAVSDK-Python/mavsdk/bin/

# removing built tools
RUN pip3 uninstall -r requirements-dev.txt -y
RUN apt-get --purge remove cmake build-essential colordiff git doxygen -y
RUN apt autoremove -y
# removing some files from MAVSDK to save some storage space.
WORKDIR /usr/src/app/
RUN rm -r ./MAVSDK/example ./MAVSDK/tools ./MAVSDK/.git ./MAVSDK-Python/examples ./MAVSDK-Python/tests

# build the image using: docker image build -t mavsdk-python-armv8:0.1 .
# run the image using: docker run --rm -it -p 127.0.0.1:4001:14540/udp 62imageid4551 sh

# To test the docker container, you can copy the telemetry.py file into a running container.
```

The 127.0.0.1:4001 is simply the port i chose to route my mavlink messages to on the host.

julianoes, 2020-03-02:
Nice, thanks for documenting all you did, that can certainly be helpful.

Regarding .deb files. We already have that for x86_64, if you look at the release assets: https://github.com/mavlink/MAVSDK/releases/tag/v0.24.0

The current debs are however without the mavsdk_server but only the library. I think a first step is to extend the debs to include the mavsdk_server and then in a second step build it as part of the dockcross arm builds.

julianoes, 2020-11-20:
This now works using statically compiled mavsdk_server binaries using musl.

closed 2020-11-20 by julianoes