mavlink/MAVSDK #999: Organize & Cleanup Receiver System & Component ID Handling

coderkalyan, 2020-03-04:
Hello, I was looking into the source code and I found quite a few FIXME's in `mavsdk_impl.cpp` regarding how different system and component IDs are handled. I created this issue so we can try to fix these issues, and to offer my thoughts on how they should be implemented as well as hear other people's ideas.

1. [L256](https://github.com/mavlink/MAVSDK/blob/master/src/core/mavsdk_impl.cpp#L256): Agreed, there are a couple things wrong here. First of all, we should not be setting the SID to a known CID, that's bad usage of the definitions. However, I don't think we should simply hard code in another value. See point 3 for my thoughts on a better approach.
2. [L258](https://github.com/mavlink/MAVSDK/blob/master/src/core/mavsdk_impl.cpp#L258): Again, this time at least we are setting the CID to a value designated for a Component ID (but incrementing it by 1? :thinking: ). This is still a fairly arbitrary value, and furthermore is hardcoded.
3. Personally, I think the `Configuration` structure is rather limiting. It forces the user to use a specific SID and CID (which is automatically decided for them) which further has the implication of only being able to successfully run 1 instance of MAVSDK at a time (well, maybe 1 per `Configuration` but you get the idea). Furthermore, the hardcoded values are set to some weird values! :thinking: 

My suggestion is: as far as API goes, start with a `void set_mavlink_id(uint8_t sid, uint8_t cid)` or similar which allows for manual user setting of the SID and CID. The current `Configuration` idea can still remain as a feature but should be refactored to something like `set_auto_mavlink_id(MavType type)` which makes more sense to me - it allows the user to have MAVSDK automatically choose an *appropriate* SID and CID based on the type of system the user is implementing with the SDK. Also, I personally feel that setting either the SID and CID or the `MavType` should be mandatory at the constructor level, but I'm not sure how others feel about this?

Now for what those auto values should be:
* For `MavType GCS`: Quoted from MAVSDK source: "Usually QGC identifies itself with sysid 255" - we should just pick a standard ID that we want to be the default MAVSDK SID, and stick to it. According to the MAVLink documentation, it is good practice to count GCS SIDs down from 255. So I'm throwing the value `245` out there, as its reasonably close to the top but leaves some room for multiple QGC/other GCS instances to be run. In reality any SID would work, and we should probably do at least some basic searching before deciding on one, to minimize conflicts (at least with existing open source GCS). The CID should be set to `MAV_COMP_ID_MISSIONPLANNER` as this is most likely what the user needs (stated in docs as "Component that can generate/supply a mission flight plan (e.g. GCS or developer API).", which is most likely the usage of MAVSDK in the context of a GCS.)
* For `MavType CompanionComputer`: This is a bit more difficult, As the FIXME says it should be autodetected. The best way to do this would probably be to read incoming `MAV_HEARTBEAT` messages until one with the CID `MAV_COMP_ID_AUTOPILOT1` is discovered, and set our SID to the source SID of that message. As for CID, `MAV_COMP_ID_PATHPLANNER` would probably be the most idiomatic, considering the companion computer is usually involved in path planning, obstacle avoidance, etc.
* For `MavType Autopilot`: As far as default values go, the existing is fine - 1 as SID and and `MAV_COMP_ID_AUTOPILOT` as CID.

4. [L263](https://github.com/mavlink/MAVSDK/blob/master/src/core/mavsdk_impl.cpp#L263): See point 3.
5. [L78](https://github.com/mavlink/MAVSDK/blob/master/src/core/mavsdk_impl.cpp#L78): This is a tough one. Could someone explain to me why we are blanket ignoring all QGC messages? Is this a QGC specific thing, or do we just not want to handle messages from other GCS? If that's the case, we should again use heartbeat to get a list of GCS based on `MAV_COMP_ID_MISSIONPLANNER` or similar, and ignore anything in that list. This seems restrictive, however, as you can't communicate with other GCS... i.e if 2 MAVSDK instances were communicating with each other. Either way, checking a specific SID is a bad idea - I for one run multiple QGC instances with different SIDs, so this wouldn't work off the bat. Would something bad happen if we didn't ignore GCS messages? Instead of "blacklisting", maybe we could "whitelist" by deciding on a per-message basis if we are capable of handling it?

These are just my thoughts - please feel free to comment and share ideas. Once we have a good idea of what we want to do, I can open a PR and implement the changes!

JonasVautherin, 2020-03-04:
Thanks for the thoughts! Here are some of mine:

> Could someone explain to me why we are blanket ignoring all QGC messages?

I could guess that this was a legacy fix (e.g. MAVSDK was connecting to the first system it could find without checking if it had an autopilot), but that it may just work now. @julianoes an opinion here?

> My suggestion is: as far as API goes, start with a void set_mavlink_id(uint8_t sid, uint8_t cid) or similar which allows for manual user setting of the SID and CID.

What about having the `Configuration` object contain the SID and CID? I could imagine having an `AutopilotConfiguration` object that can take an SID and CID, and has a reasonable default, and a `CompanionConfiguation` that would only take the CID (because it auto-detects the SID). Could that do it? I think it is the same as your `set_mavlink_id` and `set_auto_mavlink_id` suggestions, but using OOP :innocent:.

coderkalyan, 2020-03-04:
@JonasVautherin I like that idea, more than my non-OOP approach actually. To add on to that, I would suggest we have a Configuration base class, with different subclasses of it (`GCSConfiguration`, `CompanionConfiguration`). They can each have multiple constructors (one for completely autofilled values, one for setting SID and/or CID depending on the scenario/context as you mentioned. Is my understanding correct?

coderkalyan, 2020-03-05:
@JonasVautherin here's a PR that implements `Configuration`, still waiting for thoughts on what to do about QGC/ID 255 filtering. Having multiple subclasses proved very messy to implement for very limited benefit, so I just went with a single `Configuration` class with an `enum class UsageType` inside that, which provides similar functionality to the original `enum class Configuration`.

PR #1001 (2020-03-26): Implement source system and component ID handling via Configuration class
close #999.

This PR implements a a new `Configuration` class, replacing the old `enum class` of the same name. It allows for more flexibility and full control over the source system and component IDs of `Mavsdk` objects, in addition to fixing certain defaults to be more precise.

The `Configuration` class has 2 constructors - 1 which takes in the system and component id, for full manual control. The second one takes in a `UsageType` - which is an `enum class` inside `Configuration` which reimplements the same functional [... cut]
closed 2020-03-26 by julianoes