mavlink/MAVSDK #1003: No result with set_current_mission_item_async

julianoes, 2020-03-05:
There was a report that `set_current_mission_item_async` does not call the callback at all.

This needs fixing and we also need an integration tests checking it.

```
    int current = 4;
    auto prom = std::make_shared<std::promise<Mission::Result>>();
    auto future_result = prom->get_future();
    std::cout << "Setting current mission item..." << std::endl;
    mission->set_current_mission_item_async(current,
    [prom](Mission::Result result) {
        prom->set_value(result);
    });    const Mission::Result result = future_result.get(); //Wait on result    if(result == Mission::Result::SUCCESS){
        std::cout << "OK" << std::endl;
    }else{
        std::cout << "NOK" << std::endl;
    }
```

rohan2734, 2020-04-19:
This bug doesnt seem it is for a beginner

coderkalyan, 2020-07-19:
https://discuss.px4.io/t/mavsdk-clear-mission-async-callback-never-gets-called/17686 seems to be a report that `clear_mission_async` doesn't call the callback either. Maybe related?

JonasVautherin, 2020-07-20:
> Maybe related?

Yes, maybe. Would you mind linking this issue from the discuss thread? :blush:

coderkalyan, 2020-07-20:
Sure thing, will do.

julianoes, 2020-11-23:
Fixed by #1006, and covered by an integration test in 1264.

closed 2020-11-23 by julianoes