mavlink/MAVSDK #1014: PositionBody not supported by Telemetry Class?

eng-86, 2020-03-20:
hello, i have been working with MAVSDK from about month. it is great, and i have implemented many examples. when implementing a telemetry example to read the Position of a vehicle during SITL i was able to receive true data. when using requesting PositionBody i received false data.
my question is: is PositionBody  supported by Telemetry Class? due to its importance when implementing specific navigation algorithms in offboard modes, or when implementing cooperative drones...

JonasVautherin, 2020-03-20:
Can you be a bit more specific? Can you show code that you wrote, and samples of what you call "false data"?

eng-86, 2020-03-21:
Hello again, this is my working code for reading the position of multiple drones from a leader drone synchronously. I tried this code using multiple drones SITL with GAZEBO and QGC and was able to read info of 4 iris vehicles it worked great. 
`//
// Example access telemetry of multiple vehicles.
//./telemetry udp://:14540 udp://:14541 ....
//
// Author: H. Mazeh
// Author: 17/03/2020

#include <mavsdk/mavsdk.h>
#include <mavsdk/plugins/telemetry/telemetry.h>

#include <cstdint>
#include <iostream>
#include <thread>
#include <chrono>
#include <functional>
#include <future>
#include <memory>
#include <string>
#include <ctime>
#include <fstream>

using namespace mavsdk;
using namespace std::chrono;
using namespace std::this_thread;

#define ERROR_CONSOLE_TEXT "\033[31m" // Turn text on console red
#define TELEMETRY_CONSOLE_TEXT "\033[34m" // Turn text on console blue
#define NORMAL_CONSOLE_TEXT "\033[0m" // Restore normal console colour

int main(int argc, char** argv)
{
    Mavsdk dc;

    int total_udp_ports = argc - 1;

    // the loop below adds the number of ports the sdk monitors.
    for (int i = 1; i < argc; ++i) {
        ConnectionResult connection_result = dc.add_any_connection(argv[i]);
        if (connection_result != ConnectionResult::SUCCESS) {
            std::cerr << ERROR_CONSOLE_TEXT
                      << "Connection error: " << connection_result_str(connection_result)
                      << NORMAL_CONSOLE_TEXT << std::endl;
            return 1;
        }
    }

    std::atomic<signed> num_systems_discovered{0};

    std::cout << "Waiting to discover system..." << std::endl;
    dc.register_on_discover([&num_systems_discovered](uint64_t uuid) {
        std::cout << "Discovered system with UUID: " << uuid << std::endl;
        ++num_systems_discovered;
    });

    // We usually receive heartbeats at 1Hz, therefore we should find a system after around 2
    // seconds.
    sleep_for(seconds(2));

    if (num_systems_discovered != total_udp_ports) {
        std::cerr << ERROR_CONSOLE_TEXT << "Not all systems found, exiting." << NORMAL_CONSOLE_TEXT
                  << std::endl;
        return 1;
    }

    std::vector<uint64_t> system_vector = dc.system_uuids();
    std::shared_ptr<Telemetry> telemetry[system_vector.size()];

    for (unsigned i=0; i<system_vector.size(); i++){
        telemetry[i] = std::make_shared<Telemetry>(dc.system(system_vector[i]));

        // Set position update rate to 1 Hz.
        const Telemetry::Result set_rate_result1 = telemetry[i]->set_rate_position(1);
        if (set_rate_result1 != Telemetry::Result::SUCCESS) {
            // handle rate-setting failure (in this case print error)
            std::cout << "Setting rate failed telemetry:"<< i+1 << " " << Telemetry::result_str(set_rate_result1) << std::endl;
        }
    }

    Telemetry::Position position1[system_vector.size()];        
    //Telemetry::PositionBody position2[system_vector.size()];

    while(1) {
        for (unsigned i=0; i<system_vector.size(); i++){
            position1[i] = telemetry[i]->position();
            std::cout << "Altitude" << i+1 <<  " " << position1[i].relative_altitude_m << " m" << std::endl
                      << "Latitude" << i+1 <<  " " << position1[i].latitude_deg << std::endl
                      << "Longitude" << i+1 <<  " " << position1[i].longitude_deg << std::endl;
        }

        /*for (unsigned i=0; i<system_vector.size(); i++){
            position2[i] = telemetry[i]->position();
            std::cout << "x" << i+1 <<  " " << position2[i].x_m << std::endl
                      << "y" << i+1 <<  " " << position2[i].y_m << std::endl
                      << "z" << i+1 <<  " " << position2[i].z_m << std::endl;
        }*/
        sleep_for(seconds(1));
    }

    return(0);
}
`
My future purpose research  is to implement PSO, GA... online algorithms for path planning of cooperative UAVs, and i need to access the body frame position values. I could not access it since I did not found functions for telemetry position body update and telemetry position body data request as for position we have set_rate_position() and position().
thanks for help

eng-86, 2020-03-21:
Hello, for more clearance global position can be read, but body position cannot. May be NED position also??

julianoes, 2020-03-21:
Ok, and `position_velocity_ned()` does not work for you?

https://github.com/mavlink/MAVSDK/blob/12f1cb9b07cd8d2863fa1577f1082971e731aab4/src/plugins/telemetry/include/plugins/telemetry/telemetry.h#L750-L755

eng-86, 2020-03-26:
@julianoes  sorry for late reply, position_velocity_ned() worked. I will perform a function that transforms from NED to BODY frame.
thanks

closed 2020-03-26 by julianoes