mavlink/MAVSDK #1018: Instructions how to add functionality

julianoes, 2020-03-22:
Once the new autogeneration of the C++ part is done we should add documentation on how functionality can be added.

coderkalyan, 2020-03-26:
What's the new autogeneration of the C++ part? Just curious.

JonasVautherin, 2020-03-27:
@coderkalyan: The goal is to auto-generate the whole gRPC logic from the proto files, and also the C++ headers. See here for the action plugin: https://github.com/mavlink/MAVSDK/pull/1017

coderkalyan, 2020-03-29:
@JonasVautherin thats really cool! Can I help port anything else to autogenerated?

JonasVautherin, 2020-03-31:
I don't think so, but thanks :slightly_smiling_face:. Would be nice to make some tests once they are there, though, to see if that does not break functionality! :innocent:

julianoes, 2020-11-20:
This is being covered in https://github.com/mavlink/MAVSDK-docs/pull/219.

closed 2020-11-20 by julianoes