mavlink/MAVSDK #1026: Set drone yaw in a MissionItem

depauwjimmy, 2020-04-01:
I have been trying Mission lately to take exactly the same picture again automatically.
It seems to work fine except for the drone yaw which changes somewhat randomly.
There is no option in MissionItem for me to specify a yaw or anything close to an orientation to make sure both him and the gimbal are facing the same way as the photo before.

Is it intentional ? How can i be sure to orient the camera exactly like when the first photo was taken?

Thanks for you insight

hamishwillee, 2020-04-02:
Looking at [MissionItem](https://mavsdk.mavlink.io/develop/en/api_reference/classmavsdk_1_1_mission_item.html) it seems you are correct - it has gimbal yaw, but not vehicle yaw (which may differ).
@julianoes any reason not to support setting ROI too?

----

That said, this should not be "somewhat random". The way this works in MAVLink is that the [waypoint-like-items](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_WAYPOINT) have a yaw field. If the yaw is explicitly specified then that will be used. If the yaw is a NaN the vehicle will point towards whatever the system setting is (that is set using parameters, and default to "the next waypoint". 
The yaw can also be affected by the gimbal and ROI. 

If you were working in raw mavlink then I'd suggest calling the command to set an ROI. Not sure what to do now, but Julian probably will be!

julianoes, 2020-04-02:
@hamishwillee so this would be for vehicles without gimbals which literally have to turn to take a picture? I have many open question when it comes to ROI, modes of yaw during a mission, gimbal ROI, etc. To me it sounds like you need to be lucky in combining all these things to get it to work, and with MAVSDK we try to hide that sort of complexity and give you something (and sometimes only one thing) that works.
So yes it would be nice to support all this but we need understand all these aspects.

hamishwillee, 2020-04-02:
@julianoes I think you should just look at the original question, rather than my response :-). 

The takeaway for me is not a specific answer to that question, but an observation that  the `MissionItem` should support yaw too ... and this might help the specific case. 

ROI would help too. I don't want to hijaack this question with the broader discussion.

depauwjimmy, 2020-04-06:
I guess the 'somewhat random' is yaw toward waypoint right now since the movement is always the same. I just had no idea what was the referential.
It's totally not helping me though, i really need to control the yaw by myself in code.

JonasVautherin, 2020-04-06:
I am not sure I understand the use-case :confused:. Right now, you can control the yaw of the camera, which works, while the drone will always keep looking at the next waypoint. Correct?

And you would like to be able to control both the yaw of the gimbal and the yaw of the drone, for some reason?

depauwjimmy, 2020-04-06:
The gimbal yaw is not really used, we mostly need it for pitch changes and maybe roll later.
Our drone has other sensors and stuff pointing forward which are not on the gimbal. Instead of overloading the gimbal with too much stuff we find it a lot easier to control the drone yaw.
I could offset the gimbal so its facing the same thing as before but i have no idea what the drone yaw will be when reaching the waypoint so i can't give a good value when creating the mission item.

mwshafer, 2020-04-27:
I believe this issue overlaps entirely with [issue #909](https://github.com/mavlink/MAVSDK/issues/909).

JonasVautherin, 2020-04-28:
Ok, let's close this one then. @depauwjimmy: feel free to reopen if you believe it is a different issue :blush:.

closed 2020-04-28 by JonasVautherin