mavlink/MAVSDK #1031: mavlink_mission_item_to_mission_item_indices never cleared

Katawann, 2020-04-07:
_mission_data.mavlink_mission_item_to_mission_item_indices is never cleared which means we might have more mavlink_mission_item_to_mission_item_indices than we want if we download a list of items smaller than the previous one

https://github.com/mavlink/MAVSDK/blob/8afd4640cebf4fae242a9ff9650d354cebab2f48/src/plugins/mission/mission_impl.cpp#L241

julianoes, 2020-04-08:
Thanks for the report.

coderkalyan, 2020-05-18:
What's needed to fix this? Is it simply a matter of clear()ing the list before we start inserting into it?

Katawann, 2020-05-18:
yes only that hehe. I'm not working with the last version of MAVSDK and I don't have the time to upgrade for now with my job but yes if you clear it everything is fine !

coderkalyan, 2020-05-19:
Okay, if it's just clearing I can try to get a PR for this soon.

PR #1107 (2020-05-26): Fix mission indices
This is the follow up to @roangel's PR #1105.

Also fixes #1031.
closed 2020-05-26 by julianoes