mavlink/MAVSDK #1033: Timesync adjusting outgoing message timestamps

jkflying, 2020-04-09:
The usual method for timesync with mavlink is that outgoing messages are sent with the sender's clock source, and the receiver should translate it into their own time frame. However, we've seen now with MAVSDK that after an initial timesync exchange it adjusts the clock source for _outgoing_ messages. This causes massive confusion on the flight controller side, since it was trying to adjust timestamps on incoming messages to what the clock source on MAVSDK was _before_ it did the timesync. This might be the root cause of #960 ? It can also cause instability in both timesync algorithms since they try to estimate each other when the offset is constantly changing, instead of estimating a single offset which should be mostly constant.

To fix this, outgoing messages should be timestamped with a consistent clock source (preferably CLOCK_MONOTONIC_RAW, ie something that isn't affected by time jumps or temporary clock slewing on the host system), and only incoming messages should be adjusted via the timesync message.

FYI @irsdkv , @JonasVautherin suggested I tag you since you contributed this. Do you see any problems here? Would you have time to fix this?

Thanks @DanielePettenuzzo for the investigation.

irsdkv, 2020-04-09:
In our case (using Autopilot time in MoCap system messages) we have not any problems with that so I can't say anything more related to that.

But in my mind current MAVSDK tymesync mechanism (https://github.com/mavlink/MAVSDK/pull/890) can't cause these problems. Similar behavior was [implemented in ROS](https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/sys_time.cpp).
But this is just my guess and I can be wrong.

jkflying, 2020-04-09:
Reading through your PR I don't see a problem there, everything looks good. However it looks like it has been changed since then: https://github.com/mavlink/MAVSDK/blob/8afd4640cebf4fae242a9ff9650d354cebab2f48/src/core/timesync.cpp#L42
It now responds with the offset from autopilot time, not system time, which is exactly the problem we now see on flight control side.

irsdkv, 2020-04-09:
> it has been changed since then:

Yes it is related to https://github.com/mavlink/MAVSDK/pull/934

irsdkv, 2020-04-09:
Something more about this solution:

1. Source:
https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/sys_time.cpp#L310

2. I think it can be a bad solution if `ros::Time::now()` is host-machine time, not local adjusted clocks.

jkflying, 2020-04-09:
So I can understand that it might be a problem if we want to have MAVSDK at the same time as MAVROS running, because each of these is only designed to sync with a single external source.

What #934 does breaks the assumption of the timesync module in the Firmware, that the MAVSDK clock has constant offset, since Firmware sends at 10Hz, so gets responses and estimates the MAVSDK time offset before the MAVSDK has the firmware offset and shifts the clock in the first big step.

Perhaps one way (it feels a bit hacky) is to not respond to external timesync messages until the first timesync has completed with the autopilot. So [this line ](https://github.com/mavlink/MAVSDK/blob/8afd4640cebf4fae242a9ff9650d354cebab2f48/src/core/timesync.cpp#L45) needs to have something like `&& timesync_set` which is set true when the autopilot time is actually updated.

The 'proper' solution, a lot more work, is to fix the MAVSDK/Firmware/MAVROS to have multiple timesync offsets, based on component ID.

PR #1034 (2020-04-09): Don't respond to timesync until time known
The timesync protocol expects clock offsets to be constant (or at least have constant skew), however the MAVSDK has an initial big change when syncing with the flight controller. This can confuse other systems on the mavlink network.

This prevents the MAVSDK from responding to other timesync messages until its own timesync with the autopilot time has been determined.

This is a ~~fix~~ cover up for #1033 . The real fix is to allow the flight controller, MAVROS, etc to have multiple different external systems timesynced at the same time.

FYI @irsdkv  [... cut]
closed 2020-04-09 by JonasVautherin