mavlink/MAVSDK #1039: Support for multi-par statustext messages

julianoes, 2020-04-14:
We should add support for multipart statustext messages as specified in https://github.com/mavlink/mavlink/pull/1295.

krady21, 2020-06-11:
I have a few questions. I assume that the statustext messages can arrive out-of-order, right? In that case, do you think that reordering should be done on the fly or only when we get the last message? Another important aspect is what do we do if we don't receive the last message (the one null terminated). Do we implement some timeout mechanism and log just what has been received up until that point in time?

julianoes, 2020-06-11:
Good questions. I don't think we try very hard to deal with out-of-order in MAVLink in general, unless for things that are implemented to be guaranteed such as commands or a mission upload.

While a debug output is important the message is just a stream and you can potentially miss it all, that's just currently the state until we have the [events interface](https://docs.google.com/document/d/18qdDgfML97lItom09MJhngYnFzAm1zFdmlCKG7TaBpg) which deals with that using an events counter, if I remember it correctly.

Anyway, I would suggest to parse and display whatever arrives (in order) and just omit whatever does not. I think that will already bring value for most cases.

krady21, 2020-06-11:
Great. Thanks for the input. I see that calibration and telemetry also have implementations of process_statustext. Should we change those too?

julianoes, 2020-06-12:
> Should we change those too?

Yes that would make sense. And there an implementation that waits for chunks if the id is set until it prints them out. and a timeout in case the final chunk does not arrive.

PR #1127 (2020-09-25): Add support for statustext sequences
Fixes #1039
closed 2020-09-25 by julianoes