mavlink/MAVSDK #1056: Rename all leftover `dc`, `dcsdk` etc. to `mavsdk`

julianoes, 2020-04-23:
Back when MAVSDK was named differently we also called the instance of `Mavsdk` different. Leftovers are still sprinkled all over the integration tests and examples. It would be nice to get them cleaned up.

Siddharth1698, 2020-05-06:
I am new to working on open source. If this is an manual and easy task just tell me ill work on it.

julianoes, 2020-05-06:
Sure, that would be great. I would quickly wait until https://github.com/mavlink/MAVSDK/pull/1076 and https://github.com/mavlink/MAVSDK/pull/1075 are merged to avoid conflicts though.

Here is an example:
https://github.com/mavlink/MAVSDK/blob/242437c1b29d2e78cd60ca9383fda8d7b6858a6b/src/integration_tests/mission_cancellation.cpp#L26

It should just be `Mavsdk mavsdk;`.

Siddharth1698, 2020-05-06:
Cool. Got this. I should do this completely on all the files right.

JonasVautherin, 2020-05-06:
> Cool. Got this. I should do this completely on all the files right.

Ideally, yes. If it's too much work, it can be split in smaller chunks, too :blush: (like integration tests first, then examples, etc). Your call :wink:!

Siddharth1698, 2020-05-06:
No prob. Got it.. :+1:

PR #1270 (2020-12-05): Cleanup
Best reviewed commit by commit.

Closes #1055, closes #1056.
closed 2020-12-05 by julianoes