mavlink/MAVSDK #1101: Mission generation should use NAV_DELAY instead of sending two waypoints on the same location

LorenzMeier, 2020-05-15:
From @julianoes:
> The reason MAVSDK sends the same lat/lon twice in a row is because it adds a LOITER_TIME right after a NAV_WAYPOINT. The right approach would be to use NAV_DELAY [...].

Which is implemented in PX4 for quite some time (I verified). The reason PX4 rejects now waypoints that are on top of each other is because the path between those waypoints is not defined (zero distance). Often we however need to know where the vehicle is coming from geometrically and allowing N waypoints on the same location would force us to traverse the mission multiple times and quite a bit forward and backward, increasing the complexity of the autopilot in terms of memory, execution time and also in terms of our ability to QA and certify the code.

So in short: This is undesirable in principle and navigation commands should not exist in the same physical location one after the other.

julianoes, 2020-05-15:
> So in short: This is undesirable in principle and navigation commands should not exist in the same physical location one after the other.

If this is a hard rule it should also be added to the MAVLink standard.

dayjaby, 2020-05-15:
There are a couple of scenarios that cannot be done if there is such a hard rule:
(PREVIOUS WAYPOINT), WAYPOINT, DELAY, IMAGE_START_CAPTURE, LOITER_UNLIMITED for example if the waypoint and loiter_unlimited have the same position. We could prevent this reverse transversing if all the mission items in such a sequence would have the same PREVIOUS WAYPOINT assigned.

julianoes, 2020-05-18:
@dayjaby I suppose for this case we should be using WAYPOINT, DELAY, IMAGE_START_CAPTURE, DELAY, but maybe that's not possible given the current implementation.

coderkalyan, 2020-09-23:
If NAV_DELAY is supported in PX4, is there something preventing this from being implemented? Or can I take a crack at it (time permitting)?

julianoes, 2020-09-25:
@coderkalyan feel free :)

PR #1210 (2020-10-29): Mission: use NAV_DELAY instead of LOITER_TIME
Closes #1101, replacing MAV_CMD_NAV_LOITER_TIME with MAV_CMD_NAV_DELAY. Additionally, added checks for whether or not speed actually changed when inserting MAV_CMD_DO_CHANGE_SPEED mission items.
closed 2020-10-29 by julianoes