mavlink/MAVSDK #1103: Use std::unordered_map instead of std::map?

roangel, 2020-05-20:
I am not very sure of this, but wouldn't it be better (O(1) vs O(log(n)) lookup time) if we used `std::unordered_map` instead of `std::map` in these and similar cases?
https://github.com/mavlink/MAVSDK/blob/578c70c37cf7faf6603431dfba5d073c343b9db0/src/core/call_every_handler.h#L36
https://github.com/mavlink/MAVSDK/blob/578c70c37cf7faf6603431dfba5d073c343b9db0/src/core/system_impl.h#L320

julianoes, 2020-05-20:
Totally! I think at the time of writing I wasn't aware of `std::unordered_map` :man_facepalming:.

roangel, 2020-05-20:
Nice, I can take care of changing it. I was also thinking that this part:
https://github.com/mavlink/MAVSDK/blob/578c70c37cf7faf6603431dfba5d073c343b9db0/src/core/mavlink_message_handler.h#L29

Can also be an `std::unordered_map`, with the msg id as key? or even a custom hash of the msg id and the cookie? Would make the lookup time be so much faster (O(n) vs O(1))

julianoes, 2020-05-20:
Yes, I was looking at that today. The `msg_id`s can be duplicate but the `cookie` is unique, however, the unregistering won't happen very often.

Anecdotally, I've heard that just looping through a relatively short `std::vector` can be quicker in practice than having to do lookups in a map, so I would try to profile it before changing.

roangel, 2020-05-20:
> Anecdotally, I've heard that just looping through a relatively short `std::vector` can be quicker in practice than having to do lookups in a map, so I would try to profile it before changing.

Yes, I have also heard that it's because of the overhead of building the hash table behind `unordered_map`. Not sure at which quantities it starts paying off.

closed 2020-05-25 by julianoes