mavlink/MAVSDK #1106: [C++] how to stream position information from UAV

KrisRadowski, 2020-05-23:
Hey,

I want to make a stream of Position to use for eg. turning it into TargetLocation, so one drone will go after other. How to do that?

hamishwillee, 2020-05-24:
Have you considered just using the existing FollowMe mode? https://mavsdk.mavlink.io/develop/en/examples/follow_me.html

That only works for multicopters, but almost certainly does what you want if the multicopter is your desired vehicle.

hamishwillee, 2020-05-24:
PS The area is primarily for bugs/enhancement requests. For "help" see https://mavsdk.mavlink.io/develop/en/#getting-help

closed 2020-05-24 by hamishwillee

julianoes, 2020-05-25:
You can use offboard control to command a drone to a place:
https://mavsdk.mavlink.io/develop/en/api_reference/classmavsdk_1_1_offboard.html
or goto:
https://mavsdk.mavlink.io/develop/en/api_reference/classmavsdk_1_1_action.html#classmavsdk_1_1_action_1afb3546fa994357e491816f2032716818