mavlink/MAVSDK #1108: Add MAVSDK to Vcpkg

edalm, 2020-05-25:
[Vcpkg](https://github.com/Microsoft/vcpkg) is a package manager for C++ with many interesting features. It is based on CMake, supports multiple architectures, provides update mechanisms and integrates with any user project using CMake as its build manager.

Are there plans to add MAVSDK to Vcpkg? It would provide great benefits to developers using MAVSDK in their projects.

julianoes, 2020-05-26:
That would be a very good idea. I would have to read more about how it all works though. If you can give me pointers or instructions how to proceed that would be helpful.

edalm, 2020-05-26:
Thank you for considering this suggestion.

**Vcpkg Overview**

Personally, I have been using Vcpkg for some time as an end user. I find it a very useful C++ package manager, which provides an easy mechanism to search, install and update libraries for any supported architecture. This is especially useful when developing on Windows, since libraries can be automatically downloaded, compiled and installed on a single directory, which can then be targeted with CMake rules.

In summary, Vcpkg downloads the library's source code from a public URL (e.g., GitHub repo), compiles the code for the chosen architecture, and installs the library on the predefined Vcpkg directory containing all installed libraries.

Then, developers who want to use these libraries in their projects only need to add Vcpkg's CMake toolchain by appending the following flag to CMake:
"-DCMAKE_TOOLCHAIN_FILE=C:/vcpkg/scripts/buildsystems/vcpkg.cmake"

**Contribute MAVSDK to Vcpkg**

Vcpkg's GitHub page contains instructions on how to add a new library to Vcpkg.
https://github.com/microsoft/vcpkg/blob/master/CONTRIBUTING.md#new-package-guidelines

Essentially, the project maintainer has to submit a PR on Vcpkg with the files required by Vcpkg. I have never submitted a project to Vcpkg, but from my understanding Vcpkg needs two mandatory files and optional patches.

The mandatory files are:
1. ```"CONTROL"```: with the library's information.
2. ```"portfile.cmake"``` with the rules on how to download, compile and install the library.

Optionally, patches may be provided in order to fix temporary issues either with Vcpkg or the project itself.

The list of libraries currently supported by Vcpkg are listed here:
https://github.com/microsoft/vcpkg/tree/master/ports

JonasVautherin, 2020-05-27:
Sounds nice! Wouldn't we need to check that our dependencies are available on Vcpkg, too? I guess we won't add a superbuild there, will we?

edalm, 2020-05-28:
I'm not sure, but I guess every package submitted to Vcpkg should have its dependencies there too. This allows Vcpkg to automatically download and compile the dependencies required by the package to be installed.

The list of libraries currently supported in Vcpkg can be seen here (1000+):
https://github.com/microsoft/vcpkg/tree/master/ports

JonasVautherin, 2020-05-28:
> EDIT: Therefore, MAVSDK wouldn't be responsible for building third_party libraries, as Vcpkg would do it automatically. These dependencies could then be linked using CMake's find_package().

That's already how MAVSDK works (i.e. all the dependencies are found with `find_package`). By default, we have `-DSUPERBUILD=ON`, which builds the dependencies at configure time and then sets `CMAKE_PREFIX_PATH` accordingly. But if you build with `-DSUPERBUILD=OFF`, it will get all the dependencies from your system.

So the way there would be to try to build with `-DSUPERBUILD=OFF` and to see if all of our dependencies are available in Vcpkg already. You can find them [here](https://github.com/mavlink/MAVSDK/tree/develop/third_party), but note that some of them are used for `mavsdk_server` (a.k.a. "backend") exclusively, as can be seen [here](https://github.com/mavlink/MAVSDK/blob/develop/third_party/CMakeLists.txt#L19-L25). Which means:

Dependencies for MAVSDK-C++:
* jsoncpp
* tinyxml2
* zlib
* curl

Dependencies for `mavsdk_server`:
* All the dependencies of MAVSDK-C++
* libressl (or openssl or boringssl)
* cares
* protobuf
* absl
* grpc

I would start with MAVSDK-C++ (i.e. `-DBUILD_BACKEND=OFF -DSUPERBUILD=OFF`). @edalm can you check if Vcpkg has `jsoncpp`, `tinyxml2`, `zlib` and `curl`?

edalm, 2020-05-29:
I skimmed through Vckpg docs and it seems to be well documented.
https://github.com/microsoft/vcpkg/tree/master/docs

**Dependencies**

According to this [page](https://github.com/microsoft/vcpkg/blob/master/docs/maintainers/control-files.md), the dependencies of a given project should be listed in the ```CONTROL``` file, which has a syntax based on Debian's Control format. For instance, the curl library in Vcpkg depends on zlib. Thus, the curl's CONTROL file has the following line: ```Build-Depends: zlib```.

Regarding the dependencies of MAVSDK-C++, they are all supported in Vcpkg:

* ```jsoncpp```: https://github.com/microsoft/vcpkg/tree/master/ports/jsoncpp
* ```tinyxml2```: https://github.com/microsoft/vcpkg/tree/master/ports/tinyxml2
* ```zlib```: https://github.com/microsoft/vcpkg/tree/master/ports/zlib
* ```curl```: https://github.com/microsoft/vcpkg/tree/master/ports/curl

**Portfile**

The instructions for building and installing MAVSDK should be declared in the ```portfile.cmake```.
Vcpkg provides CMake functions to manage this process, which are documented here:
https://github.com/microsoft/vcpkg/blob/master/docs/maintainers/portfile-functions.md

JonasVautherin, 2020-05-29:
Awesome! I think you should try to go for it, then :slightly_smiling_face:. I would personally try to use Vcpkg to build MAVSDK first (as opposed to writing the portfile to make a package out of MAVSDK):

1. Clone MAVSDK
2. Setup your Vcpkg to get the dependencies (`jsoncpp`, `tinyxml2`, `zlib`, `curl`)
3. Build MAVDSK with the toolchain, as per your instructions above (don't forget to disable the superbuild). So something like:

```cmake
cmake -DCMAKE_TOOLCHAIN_FILE=C:/vcpkg/scripts/buildsystems/vcpkg.cmake -DSUPERBUILD=OFF -Bbuild -S.
cmake --build build
```

If this builds, it means that MAVSDK was built using the dependencies from Vcpkg. Next step would then be to write the portfile to make a package out of MAVSDK, I guess.

edalm, 2020-05-29:
Thank you for the pointers. I will try that and update this post when I have results.

tpool2, 2020-08-06:
I greatly appreciate all the work that's been done! Are there any updates to this issue?

edalm, 2020-08-06:
Unfortunately, I've been a little busy in the last few months and I haven't been able to make much progress on this.

I remember that a few months ago I tried to build MAVSDK using Vcpkg, as suggested by Jonas, but I had a few issues. I haven't tried to repeat the process using the updated MAVSDK and Vcpkg.

On the other hand, it seems Microsoft is designing a new declarative method (using manifests) for Vcpkg, in order for users to specify the libraries used in their projects in a source-controlled file, as well as for maintainers to publish their own libraries. The draft is available on Vcpkg's GitHub:
https://github.com/microsoft/vcpkg/blob/master/docs/specifications/manifests.md

This might simplify the effort of submitting MAVSDK on Vcpkg, although I haven't tried it yet.

edalm, 2020-08-15:
Despite my interest in making this contribution, I do not know when I will have the time to work on this.

Therefore, my suggestion is to close this issue for now. If I make some progress on this issue, I can reopen it to share the details with you.

Thank you for the support and feedback provided.

closed 2020-08-15 by edalm