mavlink/MAVSDK #1109: attitude_euler bug

mathias-h, 2020-05-26:
I have noticed some trouble with the "attitude_euler" function.

in both examples the quad is just standing still.

example:
`
telemetry->subscribe_attitude_euler([](Telemetry::EulerAngle angle) {
        std::cout << angle << "\n";
});
`
output:
`
euler_angle:
{
    roll_deg: 0.901273787021637
    pitch_deg: -1.59999907016754
    yaw_deg: -31.9563980102539
}
euler_angle:
{
    roll_deg: 114.56241607666
    pitch_deg: 29.9334468841553
    yaw_deg: -33.3854217529297
}
euler_angle:
{
    roll_deg: 0.90143883228302
    pitch_deg: -1.60016667842865
    yaw_deg: -31.9572525024414
}
euler_angle:
{
    roll_deg: 0.901572823524475
    pitch_deg: -1.60093307495117
    yaw_deg: -31.9589595794678
}
euler_angle:
{
    roll_deg: 114.562004089355
    pitch_deg: 29.9333267211914
    yaw_deg: -33.3893890380859
}
euler_angle:
{
    roll_deg: 0.902443706989288
    pitch_deg: -1.60160052776337
    yaw_deg: -31.959451675415
}
euler_angle:
{
    roll_deg: 0.902292251586914
    pitch_deg: -1.60183465480804
    yaw_deg: -31.9599742889404
}
euler_angle:
{
    roll_deg: 114.561882019043
    pitch_deg: 29.9340591430664
    yaw_deg: -33.3904113769531
}
euler_angle:
{
    roll_deg: 0.902660369873047
    pitch_deg: -1.60230326652527
    yaw_deg: -31.9612464904785
}
euler_angle:
{
    roll_deg: 114.561477661133
    pitch_deg: 29.9352607727051
    yaw_deg: -33.3939323425293
}
euler_angle:
{
    roll_deg: 0.902866005897522
    pitch_deg: -1.60294258594513
    yaw_deg: -31.9626407623291
}
euler_angle:
{
    roll_deg: 0.90369713306427
    pitch_deg: -1.60316836833954
    yaw_deg: -31.9617881774902
}
euler_angle:
{
    roll_deg: 114.561592102051
    pitch_deg: 29.934398651123
    yaw_deg: -33.3930320739746
}
euler_angle:
{
    roll_deg: 0.902268767356873
    pitch_deg: -1.60252797603607
    yaw_deg: -31.9614353179932
}
euler_angle:
{
    roll_deg: 0.902015566825867
    pitch_deg: -1.60285496711731
    yaw_deg: -31.9615859985352
}
euler_angle:
{
    roll_deg: 114.561660766602
    pitch_deg: 29.9346885681152
    yaw_deg: -33.3923721313477
}
euler_angle:
{
    roll_deg: 0.902102828025818
    pitch_deg: -1.60208475589752
    yaw_deg: -31.9615478515625
}
euler_angle:
{
    roll_deg: 0.901788949966431
    pitch_deg: -1.60183584690094
    yaw_deg: -31.9628047943115
}
euler_angle:
{
    roll_deg: 114.56143951416
    pitch_deg: 29.937744140625
    yaw_deg: -33.3936347961426
}
euler_angle:
{
    roll_deg: 0.902568221092224
    pitch_deg: -1.60193073749542
    yaw_deg: -31.9638233184814
}
euler_angle:
{
    roll_deg: 114.561363220215
    pitch_deg: 29.9371948242188
    yaw_deg: -33.394416809082
}
euler_angle:
{
    roll_deg: 0.903651833534241
    pitch_deg: -1.60192632675171
    yaw_deg: -31.9637641906738
}
euler_angle:
{
    roll_deg: 0.903929471969604
    pitch_deg: -1.60198318958282
    yaw_deg: -31.9647083282471
}
`

When i use the regular attitude message, it works fine.

example:
`
mavlink_passthrough->subscribe_message_async(MAVLINK_MSG_ID_ATTITUDE, [&](mavlink_message_t message) {
        mavlink_attitude_t attitude;
        mavlink_msg_attitude_decode(&message, &attitude);

       std::cout << attitude.pitch << "deg " << attitude.roll << "deg\n";
});
`
output:
`
-0.0117168deg 0.0117473deg 
-0.0117303deg 0.011718deg 
-0.0117192deg 0.0117017deg 
-0.0116752deg 0.0116813deg 
-0.0116182deg 0.0116685deg 
-0.0115927deg 0.011668deg 
-0.0115764deg 0.011668deg 
-0.0115786deg 0.0116677deg 
-0.0115909deg 0.0116682deg 
-0.0116277deg 0.0116809deg 
-0.0116537deg 0.0116645deg 
-0.0116414deg 0.0116718deg 
-0.0116325deg 0.0116635deg 
-0.0115993deg 0.0116332deg 
-0.0115136deg 0.0115944deg 
-0.0114356deg 0.011561deg 
-0.0113072deg 0.011516deg 
-0.01118deg 0.0114664deg
`

julianoes, 2020-05-26:
Oh, that's not good! Thanks for reporting.

julianoes, 2020-05-26:
One thing. Are you sure that you don't have two vehicles sending an attitude? E.g. a real one and a SITL instance broadcasting in the network?

And FYI the attitude is assembed from either `ATTITUDE` or `ATTITUDE_QUATERNION` messages, so either something is somewhere wrong in the implementation (probably in MAVSDK) or there are multiple sources mixed.

mathias-h, 2020-05-26:
yep, only one vehicle :)
The same setup has worked for me in the past, only difference is that i just updated MAVSDK.

mathias-h, 2020-05-26:
By the way, this was on a Pixhawk 2.1 if that makes any difference.

julianoes, 2020-05-26:
Ok thanks for the information.

julianoes, 2020-06-04:
It turns out that the conversion function was just wrong. With #1120 it might be fixed.

closed 2020-06-08 by julianoes

Bert-Tec, 2020-08-06:
Had the same issue with v0.28.0, v0.29.0 fixed it.
Thanks!

PR #1120 (2020-06-08): Unit tests for angle conversions and fix conversion
Should fix #1109.