mavlink/MAVSDK #1111: Detect waypoint too far (to close) issue before start mission

dangviethieu, 2020-05-27:
Hello all. Please help me with the question.
I uploaded the mission and get the warning:
`WARN  [navigator] First waypoint too far away: 1638 meters, 900 max.`
Then I cannot start mission because this issue.
The problem is the result of function `upload_mission_async()` and `armed()` is still `SUCCESS`, but I cannot fly.
I wonder how to detect the issue and show message before I start mission in this case?
Please help me. Thanks very much.

julianoes, 2020-05-27:
@dangviethieu this is a protection against missions set up incorrectly.

You can change or disable the protection using these two params:
https://dev.px4.io/master/en/advanced/parameter_reference.html#MIS_DIST_1WP
https://dev.px4.io/master/en/advanced/parameter_reference.html#MIS_DIST_WPS

dangviethieu, 2020-05-27:
@julianoes Yes, thank sir 🙇 
I mean that we still protect missions with two params, but how to get result is fail when upload mission in this case. The SDK should return fail result and message to user when it cannot fly.

julianoes, 2020-05-27:
The upload is valid because you might be uploading the mission at home before going to the place where you will fly.
The protection is exactly against that case when you upload a mission in one place and then try to fly in another place.

I do agree though that the the mission start command should fail. I assume it does not?

JonasVautherin, 2020-05-27:
> I do agree though that the the mission start command should fail. I assume it does not?

Given the "but I cannot fly.", I would assume that `start_mission` fails, which is expected :blush:.

> I wonder how to detect the issue and show message before I start mission in this case?

You can check that from your code: get the first waypoint, get the current drone position, and compute the distance. In the case of PX4, you can even use the `param` plugin to get the params @julianoes referenced above and act accordingly :+1:.

dangviethieu, 2020-05-28:
@julianoes @JonasVautherin Yes I got it
I think we need `start_mission` fails in this case.
And about protection is exactly against, I will check distance in my code
Thanks very about the reply 🙇‍♂️

dangviethieu, 2020-05-28:
@JonasVautherin Could we get params of PX4 from our MAVSDK code?

JonasVautherin, 2020-05-28:
> Could we get params of PX4 from our MAVSDK code?

Yes, the `param` plugin of MAVSDK is there for that :wink:

closed 2020-05-28 by julianoes