mavlink/MAVSDK #1115: Zoom Support missing (Camera/Gimbal)

petergerten, 2020-05-30:
Is there support for zoom (e.g. Yuneec E90 digital zoom or Yuneec E30Z optical zoom) already?
I could not find it so far and even on raw MAVLink level this seems to be very unclear how to do.
https://developer.yuneec.com/documentation/125/Supported-mavlink-messages does not mention it either.

julianoes, 2020-06-02:
It's currently not supported. And I'm not sure if the E30Z is supported at all.

There is a MAVLink message: [MAV_CMD_SET_CAMERA_ZOOM](https://mavlink.io/en/messages/common.html#MAV_CMD_SET_CAMERA_ZOOM) but I'm not sure if that one is already implemented on the H520 side.

Let me ping @darioxz, he might know more.

Generally I agree that it would we worthwhile to add zoom to the camera plugin.

petergerten, 2020-06-03:
@julianoes I already tried to send the MAV_CMD_SET_CAMERA_ZOOM message, but could not make it work yet.
I was thinking of sniffing the MAVLink traffic between remote control and H520 camera - but as this is probably not intended to be a secret by Yuneec I wanted to ask here :)

julianoes, 2020-06-03:
It definitely makes sense to ask here! :smile: Unfortunately, I don't remember. @dogmaphobic do you remember how to set zoom for H520?

darioxz, 2020-06-03:
@julianoes Thanks for pinging me.

@petergerten To control the optical and digital zoom factor, we are using camera parameters. I've just created a PR with the necessary camera definitions file. I haven't had the chance to test it yet, but I hope it will allow you to use the zoom functionality with the MAVSDK. Have a look at #1119.

Long-term, it would be nice to support the `MAV_CMD_SET_CAMERA_ZOOM` message in our camera firmware as well as in the MAVSDK camera plugin. I've just submitted a request for this, but can't promise anything at this point.

petergerten, 2020-06-03:
@darioxz wow, thanks a lot! I did not expect this to move forward that quick.
Now I have a problem, as I only sometimes have access to a E30Z and can't test it just yet.

I was wondering if the digital zoom capabilities that are now available with the E90 work in the same way? It seems the camera definition files do not reflect that at the moment. I will probably just give it a try :)

darioxz, 2020-06-03:
@petergerten I would assume it works the same way. I will try to obtain an updated XML for the E90 (and other cameras).

darioxz, 2020-06-03:
If it doesn't work, try the following parameter:
```
        <parameter name="CAM_ZOOM" type="uint32" control="0">
            <description>Camera Zoom</description>
        </parameter>
```
It needs to be added in the E90.xml file. I am sorry for the lousy documentation. 

... and don't forget the generator after changing the xml.

dogmaphobic, 2020-06-03:
> @dogmaphobic do you remember how to set zoom for H520?

Yup, @darioxz already responded above. It's done using parameters. Zoom support for the Yuneec cameras came before there were any zoom commands in MAVLink. For simple zoom control (zoom in/out), the commands work fine. If you need precise zoom control (change optical focal length to 75mm or digital zoom to 1.25X for example), that can only be done using parameters. I don't remember how the E90 handled zoom (it has a fixed focal length lens).

julianoes, 2020-06-04:
@darioxz would be great if you could update all xmls to what is released by Yuneec.

darioxz, 2020-06-24:
@julianoes Jep, I will try to get them. 

Does the SDK still have the capability of downloading the camera definition files directly from the camera?

julianoes, 2020-06-24:
I can download the file from http. However, we had disabled that in order to support the H520 G where the Swift app connects via MFi to the controller and http was not exposed through MFi.

The correct way to deal with this would be to use MAVLink FTP. MAVSDK supports it and also has code for the server side which could serve as inspiration for the camera side.

JonasVautherin, 2020-06-24:
> The correct way to deal with this would be to use MAVLink FTP. MAVSDK supports it and also has code for the server side which could serve as inspiration for the camera side.

So the drone would give an URI like `mavftp://`, and MAVSDK would know that it should get it using mavlink ftp, right? And the drone could still give `http://`, and MAVSDK could download it over HTTP. Is that correct?

petergerten, 2020-09-01:
@darioxz Sorry, but unfortunately I don't have access to a Z30 anymore to test it. I am going to test 'digital zoom' with a E90 though.

julianoes, 2024-05-27:
Fixed in #2094.

closed 2024-05-27 by julianoes