mavlink/MAVSDK #1116: Creating/Uploading a Survey Mission

petergerten, 2020-05-30:
Is there any way to create (or even just upload) a survey mission via MAVSDK.
If I have a QGC plan file, it seems that the importer won't work as TransectStyleComplexItem is not supported.

JonasVautherin, 2020-05-30:
[This example](https://github.com/mavlink/MAVSDK/blob/develop/examples/fly_mission/fly_mission.cpp) shows how to use the `Mission` plugin. Would that help?

petergerten, 2020-06-03:
@JonasVautherin I was having trouble to import a survey QGC plan and was confused on what exactly is supported and what not. Now I am stuck with a different issue (https://github.com/mavlink/MAVSDK-Python/issues/201), so I did not get further with this.

Is there an overview of what is supported or what is not supported when importing an QGC plan file via MAVSDK?

Jonsen94, 2020-08-06:
I checked the c-code. If I understand it correctly u can basically only use Mission-Plans from QGC which are build up from SimpleItems.
Just take a look into the plan file. It contains an "items"-Array. Everything there with a COMMAND_ID field can be used.

As a quick fix u can simple go into a plan file with a survey item. The survey item (inside the "items"-array) contains an "Items" array. Copy paste the content from the inner array into the outer one (don't forget to add correct commas) and remove the survey item. You can't change it from QGC so nicely afterwards but it will at least work with mavsdk.

petergerten, 2020-08-06:
@Jonsen94 yes, I checked the code as well - and it seems to be quite a big effort to implement support for survey missions as in TransectStyleComplexItem :-(

Jonsen94, 2020-08-06:
Are you sure? I mean, basically you could simple check if the "command" exists, if not, check if "TransectStyleComplexItem" exists. If so, check if "Items" array exists (I guess then we know we are inside a survey item?) take the "Items" array and parse it with the same for loop which parses the other "items" array. You do not have to re-implement the routings or so because they are already stored in there.

petergerten, 2020-08-06:
@Jonsen94 so the items within a TransectStyleComplexItem would be sufficient to parse the mission - and all other information is only for QGC to be show the survey configuration but not required for the internal mission representation - do I understand that correctly?

Jonsen94, 2020-08-06:
afaik yes.
If I understand the mission plan correctly the rest of the item is only for QGC so it can be loaded as a survey item.

Jonsen94, 2020-08-26:
@petergerten : PR is merged and should solve the issue now.
@JonasVautherin :  With the merged PR this issue should be resolved and might be closed ?

closed 2020-08-26 by julianoes