mavlink/MAVSDK #1117: Cross-Compile for RaspberryPi Error

domib97, 2020-06-01:
Im planning to use MAVSDK only for c++ on a Pi Zero in combination with ardupilot. I followed the instructions of: Cross compilation using dockcross and got some errors: 
[CMakeError.log](https://github.com/mavlink/MAVSDK/files/4710919/CMakeError.log)
[CMakeOutput.log](https://github.com/mavlink/MAVSDK/files/4710920/CMakeOutput.log)

Do I really need this not-so-lightweight sdk if i only plan to use c++ or will it be easier to use the 
c_library_v2 ?

JonasVautherin, 2020-06-01:
Would you mind pasting here the command line you use and which failed? Also would be nice if you could paste the actual error you got.

> in combination with ardupilot

MAVSDK is developed and tested against PX4 (trying to have as few PX4-specific logic as possible), so you may have issues running it against Ardupilot. We are of course open to contributions if you're willing to improve Ardupilot support! :wink: 

> Do I really need this not-so-lightweight sdk

Could you elaborate on what you mean by "not-so-lightweight"? MAVSDK-C++ is pretty lightweight to me, as it has very few dependencies. The language bindings (MAVSDK-Python, MAVSDK-Java, MAVSDK-Swift, ...) do rely on `mavsdk_server`, which is indeed a bit heavier because it depends on gRPC. But in your case you mention MAVSDK-C++, so I would be interested in knowing what is "not-so-lightweight" there :thinking:.

> will it be easier to use the c_library_v2

And wouldn't it be easier to start from scratch, not even relying on `c_library_v2`? I would advise you go for whatever is easier for you. In my case, I find MAVSDK-C++ lightweight, and it is much more efficient for me to use it than to rewrite the whole MAVLink logic from scratch.

hamishwillee, 2020-06-02:
FWIW @domib97 To provide some context the `c_library_v2` provides "primitives" for packing and unpacking messages into buffers for sending out over streams, for common messages and for the main dialects.

What it does not do is understand anything about the [microservices](http://mavlink.io/en/services/). These are the sequences of messages that define what it means for two systems to be connected, how to exchange mission data, how to get parameters from the underlying system, etc. 
You can write all that yourself, but it can be very hard to get right, and will probably end up being "as heavyweight" as mavsdk.

ArduPilot users tend to use Dronekit (Python) or Pymavlink (Python) for companion computer code (and more recently onboard LUA scripts). They use mavproxy for routing. Dronekit-Python in particular does much of what the SDK does, but of course is Python only. It is optimised for ArduPilot. The downside is that it is not properly supported by anyone, though it does have a reasonable community. I recommend you go talk to ArduPilot on the ArduPilot/pymavlink gitter channel to confirm this advice.

If you want to use the SDK then you will need to write some plugins if you use features where ArduPilot is not compliant with the MAVLink spec (there are a few, in parameter and mission handling), or where the spec does not properly cover any flight stack (e.g. to get the ArduPilot specific modes). You will probably want to make sure that you can connect to their simulator. 
All that is a bit of effort, but if you want to use C++ probably actually a lot less then trying to write your own API. 

If you're targetting ardupilot and you must use C/C++ then you might choose to use c_library_v2?

julianoes, 2020-11-20:
The mavsdk_server is now built statically and provided by PyPi. We have no more need to do cross compilation, or building on a Raspberry Pi. If you still require it and run into problems, please create a new issue @domib97, thanks.

closed 2020-11-20 by julianoes