mavlink/MAVSDK #1121: Preflight check in simulation??

shrit, 2020-06-07:
Hello,

I am not sure if the error is related to MAVSDK or to PX4. However, after updating PX4 to the master, I started getting this error during simulation in Gazebo.

```
[02:30:23|Warn ] command temporarily rejected (400). (mavlink_commands.cpp:172)
[02:30:23|Debug] MAVLink: critical: Preflight Fail: Gyro #0 uncalibrated (system_impl.cpp:257)
[02:30:23|Warn ] command temporarily rejected (400). (mavlink_commands.cpp:172)
[02:30:23|Debug] MAVLink: critical: Preflight Fail: Gyro #0 uncalibrated (system_impl.cpp:257)
[02:30:23|Warn ] command temporarily rejected (400). (mavlink_commands.cpp:172)
[02:30:23|Debug] MAVLink: critical: Preflight Fail: Gyro #0 uncalibrated (system_impl.cpp:257)
[02:30:23|Warn ] command temporarily rejected (400). (mavlink_commands.cpp:172)
[02:30:23|Debug] MAVLink: critical: Preflight Fail: Gyro #0 uncalibrated (system_impl.cpp:257)
[02:30:23|Debug] MAVLink: critical: Primary gyro not found (system_impl.cpp:257)
[02:30:23|Debug] MAVLink: critical: Primary gyro not found (system_impl.cpp:257)
[02:30:23|Debug] MAVLink: critical: Primary gyro not found (system_impl.cpp:257)
[02:30:23|Debug] MAVLink: critical: Primary gyro not found (system_impl.cpp:257)
[02:30:23|Debug] MAVLink: critical: system power unavailable (system_impl.cpp:257)
[02:30:23|Debug] MAVLink: critical: system power unavailable (system_impl.cpp:257)
[02:30:23|Debug] MAVLink: critical: system power unavailable (system_impl.cpp:257)
[02:30:23|Debug] MAVLink: critical: system power unavailable (system_impl.cpp:257)

```
Usually, in the simulation, we do not do a calibration. However, I am not sure if the new version of PX4 allows calibration in simulation especially that this is blocking arming and taking off completely. Any thoughts?

Regards,

JonasVautherin, 2020-06-08:
I would think this comes from PX4 :thinking:

julianoes, 2020-06-08:
Odd. What version of PX4 is that? You can e.g. paste the output of `ver all`.

I'll try to reproduce it.

julianoes, 2020-06-08:
Do you also see this problem with jMAVSim?

julianoes, 2020-06-08:
I don't see this in Gazebo so far.

Jaeyoung-Lim, 2020-06-08:
@shrit Most likely you have the wrong sitl_gazebo submodule.

Try 
```
make distclean
make submodulesupdate
```
and then rebuild sitl in master

shrit, 2020-06-08:
@julianoes, @Jaeyoung-Lim Hmm, After a set of tests, it seems to be related to my code rather than `PX4` or `MAVSDK`. Sorry for that
`PX4` version is working when testing using `commander takeoff` or using mavsdk
Since I add sensors to the sdf files after I generate them using xacro, exactly as done in `sitl_gazebo`
Still, I can not figure out where the problem coming from,

julianoes, 2020-06-08:
Maybe this? https://github.com/PX4/sitl_gazebo/pull/504

shrit, 2020-06-08:
@julianoes I figure it out.
The problem is related to the` instance_*` directories that is created when starting the simulation.
I use a local slightly modified `gazebo_sitl_multiple.sh`, which generate these `instance_*` in my build directory.
Since I updated PX4, I did not delete these local folders, I only cleaned up the build in the Fimware.
When I deleted all these old `instance_*` directories the simulation is working fine now.
I am not sure if we need to keep them after each simulation or to delete them, it seems that they are not overwritten by this script `gazebo_sitl_multiple.sh` unless if I had missed it.
I know that they have the logs, but maybe we can keep them in a folder with the date and time for each simulation, what do you think? correct me if I am wrong

dssdyx, 2020-06-13:
I solved this problem. First you need to try make px4_sitl_default gazebo.. If it can run normally, it will output some parameters in terminal. Then you should find the 'CAL_GYRO0_ID' and 'CAL_GYRO_PRIME'. It should be 2294028, but maybe yours is the same as CAL_IMU_ID. You need to change it to 2294028 in QGroundControl.

julianoes, 2020-11-20:
I'm closing this. It is a similar issue to #65.

closed 2020-11-20 by julianoes

AlexWUrobot, 2023-02-05:
I have the similar issue and the iris cannot take off vertically.
I notice that there is the acceleration bias when it is on the ground
ref: https://discuss.px4.io/t/simulation-take-off-motion-not-vertical/30705

julianoes, 2023-02-07:
@AlexWUrobot looks like your issue has been fixed.

AlexWUrobot, 2023-02-07:
Yes, but just partly solve the issue.
It is very helpful to clean the `~/.ros`
ref: https://github.com/PX4/PX4-Autopilot/issues/18299

However, it still sometimes has the [ Preflight Fail: Battery unhealthy ](https://discuss.px4.io/t/preflight-fail-battery-unhealthy/30678)

Besides, if trying machine learning  after 234 times iterations  
, the terminal of ` roslaunch px4 mavros_posix_sitl.launch` shows the below error, 

[image]