mavlink/MAVSDK #1134: Comm error with RPi connected to PX4.

ssarfati, 2020-06-22:
I have having following issues with my setup and would appreciate the support suggestions.
•	Mission Planner in ROVER mode, just used to view settings and change configuration. Connected over telemetry link to the PX4.
•	RPI-3 running w/Python 3.7 mounted on the rover.
•	Serial interface between the PX4 and RPI.
•	MAVLINK works – used this for testing the serial link
	sudo mavproxy.py --master=/dev/serial0 --baudrate 9600 --aircraft MyCopter
      o	I can communicate with the Pixhawk to arm and disarm. 
•	MAVSDK loaded via pip3 install mavsdk
•	Using mavsdk_server_linux-armv7
•	Starting mavsdk server with the following “./mavsdk_server serial:///dev/serial0:9600"
      o	Receiving the following errors. Seems to be re-sending the commands. 

pi@raspberrypi:~/MAV1 $ ./mavsdk_server serial:///dev/serial0:9600
[08:09:03|Info ] MAVSDK version: 0.28.0 (mavsdk_impl.cpp:26)
[08:09:03|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:377)
[08:09:04|Info ] Server started (grpc_server.cpp:43)
[08:09:04|Info ] Server set to listen on 0.0.0.0:44545 (grpc_server.cpp:44)
[08:09:04|Info ] Waiting to discover system on serial:///dev/serial0:9600... (connection_initiator.h:22)
[08:09:04|Debug] Component Autopilot (1) added. (system_impl.cpp:352)
[08:09:04|Warn ] sending again after 0.500717 s, retries to do: 3  (520). (mavlink_commands.cpp:224)
[08:09:05|Warn ] sending again after 1.00155 s, retries to do: 2  (520). (mavlink_commands.cpp:224)
[08:09:05|Debug] Discovered 1 component(s) (UUID: 1) (system_impl.cpp:525)
[08:09:05|Info ] System discovered [UUID: 1] (connection_initiator.h:61)
[08:09:06|Error] Error: UUID changed (system_impl.cpp:206)
[08:09:06|Warn ] Command ack 520 not matching our current command: 511 (mavlink_commands.cpp:142)
[08:09:06|Warn ] sending again after 0.504455 s, retries to do: 3  (511). (mavlink_commands.cpp:224)
[08:09:06|Error] Error: UUID changed (system_impl.cpp:206)
[08:09:06|Warn ] Command ack 520 not matching our current command: 511 (mavlink_commands.cpp:142)
[08:09:06|Debug] Falling back to Gimbal Version 1 (gimbal_impl.cpp:53)
[08:09:06|Error] Error: UUID changed (system_impl.cpp:206)
[08:09:06|Warn ] sending again, retries to do: 3  (NAV_MIN_FT_HT). (mavlink_parameters.cpp:458)
[08:09:07|Warn ] sending again after 1.01098 s, retries to do: 2  (511). (mavlink_commands.cpp:224)
[08:09:07|Warn ] Command ack 511 not matching our current command: 512 (mavlink_commands.cpp:142)
[08:09:07|Warn ] sending again after 0.504672 s, retries to do: 3  (512). (mavlink_commands.cpp:224)
[08:09:07|Warn ] sending again, retries to do: 2  (NAV_MIN_FT_HT). (mavlink_parameters.cpp:458)
[08:09:07|Warn ] Command ack 511 not matching our current command: 512 (mavlink_commands.cpp:142)
[08:09:08|Warn ] sending again after 0.504686 s, retries to do: 3  (520). (mavlink_commands.cpp:224)
[08:09:08|Warn ] sending again, retries to do: 1  (NAV_MIN_FT_HT). (mavlink_parameters.cpp:458)
[08:09:08|Warn ] Command ack 512 not matching our current command: 520 (mavlink_commands.cpp:142)
[08:09:09|Error] Error: UUID changed (system_impl.cpp:206)
[08:09:09|Warn ] sending again after 0.505059 s, retries to do: 3  (528). (mavlink_commands.cpp:224)
[08:09:09|Error] Error: Retrying failed get param busy timeout: NAV_MIN_FT_HT (mavlink_parameters.cpp:474)
[08:09:09|Warn ] Command ack 520 not matching our current command: 528 (mavlink_commands.cpp:142)

julianoes, 2020-06-23:
It sounds like you are using ArduPilot (ArduRover in this case) and not PX4.
MAVSDK is not compatible with ArduPilot (yet), see: #728.

JonasVautherin, 2020-06-23:
I think it is against PX4: "Serial interface between the PX4 and RPI."

But it is a rover :innocent:.

JonasVautherin, 2020-06-23:
@ssarfati: Actually, are you sure that you are using PX4 and not Ardupilot? It is not clear to me if Mission Planner works with PX4. Note that "PX4" is not the same thing as "Pixhawk". You could be running Ardupilot on a Pixhawk, and that would not be PX4 :blush:.

ssarfati, 2020-06-23:
Hi, Sorry for the confusion here. I have only been part of this environment for a few weeks. I did not realize that MAVSDK does not currently work with Ardupilot. Ok, I will switch over to PX4.  I just loaded QGroundControl and loaded V1.10.1 of PX4. The unit boots up, but I don't see the Airframe for Rover. What airframe should I use? The instructions indicates that it does exist. In the meantime, I will just select a quadcopter and test the RPI with the PX4.  Thanks again for your quick support!

JonasVautherin, 2020-06-24:
Ok, good to know :-). So I'm closing this for now, since it's not a connection issue between MAVSDK and PX4 :innocent:.

About the configuration for the rover, I would suggest you ask in #rover on the PX4 Slack! :blush:

closed 2020-06-24 by JonasVautherin

julianoes, 2020-06-24:
Yes, rover support is very basic in PX4, and I'm not sure how well MAVSDK works with PX4 rover.