mavlink/MAVSDK #1139: Problem running tools/generate_from_protos.sh

shakeebbb, 2020-07-08:
I'm getting the following error when I run 'tools/generate_from_protos.sh'.
I ran the configuration step before running the .sh script 'cmake -DBUILD_BACKEND=ON -Bbuild/default -H.' Anyone knows what am I missing?

-------------------------------
 Error
-------------------------------
'protoc-gen-mavsdk' not found in PATH

Make sure 'protoc-gen-mavsdk' is installed and available

julianoes, 2020-07-08:
I think this changed, can you try:
```
pip3 install --user protoc-gen-mavsdk
```

JonasVautherin, 2020-07-08:
It was called `protoc-gen-dcsdk`, but now it is `protoc-gen-mavsdk` indeed. The error message says that you need to install it, as @julianoes suggests above.

PR #1140 (2020-07-08): Slight improvements to generate_from_protos.sh
This mainly adds a hint about how to install `protoc-gen-mavsdk`, plus two small cleanups, see commits.

Closes  #1139.
closed 2020-07-08 by JonasVautherin