mavlink/MAVSDK #1141: Serial connection error with mavsdk_server: "No such file or directory (serial_connection.cpp:87"

nithilan4, 2020-07-12:
I tried to connect to my pixhawk with a Holybro 100mW Transceiver Telemetry Radio Set V3 (915Mhz).
I started mavsdk_server with ./mavsdk_server -p 50051 serial:///dev/ttyUSB0
and the error
`
[03:17:47|Info ] Server started (grpc_server.cpp:43)
[03:17:47|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:44)
[03:17:47|Info ] Waiting to discover system on serial:///dev/bus/usb/001/002:57600... (connection_initiator.h:22)
[03:17:47|Error] open failed: No such file or directory (serial_connection.cpp:86)
[03:17:47|Error] Connection failed: Connection error (connection_initiator.h:48)
`
shows

I'm on ARM.

JonasVautherin, 2020-07-12:
It seems like `/dev/ttyUSB0` does not exist. Are you sure that it does?

nithilan4, 2020-07-12:
oh that's weird, it doesn't. could that be the problem?

i have super weird installation. i installed termux, then ubuntu on termux, then built the core and server

nithilan4, 2020-07-12:
i just realized what the issue was, i am not rooted and i don't think it could find it. i misread the error and thought that serial_connection wasn't found.

closed 2020-07-12 by nithilan4