mavlink/MAVSDK #1153: How to setup a VTOL takeoff mission with mission_raw?

notoriou5, 2020-07-27:
Hello everyone, 

I am trying to achieve a VTOL takeoff mission with the navigation setpoint triplet uploaded beforehand. For my specific task it is important to know the next waypoint while performing the transition. The two options I found working on the QGCS are as following:

**What I am trying to achieve:**
Option 1: perform a standard MC takeoff followed by a transition to FW
[image]

Option 2: perform a Vtol takeoff command
[image]

**Solution with mission and action commands:**
[image]

If I try to achieve my goal with the currently implemented mission and action commands it looks like shown above. The problem there is, that the vtol will not turn towards the next waypoint before performing the transition. It will just accelerate in whatever heading it is after takeoff. 
For now, lets stick with option 1 and try to achieve this with the mission_raw.

**Setup Issues for mission_raw:**
 It seems like creating a MissionItem with 3 entries should be enough to solve this task, but I am getting stuck filling my upload_mission(). I tried both options (async and normal but both seem to be _undefined reference_. I build on top of the autopilot_tester in px4

That's the setup ()
[image]

That's the function:
[image]
And terminal output:
[image]

Since the setup for my mission_raw is the same as the normal mission, I don't realy understand the _undefined reference_ problem.

**Questions/ Further Goals**

1. Is the way with mission_raw the "best" way to follow to achieve my goal?
2. One of the fail conditions would be an excessive deviation from the planned ground truth. How would you implement this?

julianoes, 2020-07-27:
Until we support VTOL in the mission plugin, the mission_raw approach makes sense.

The undefined reference problem is easy,  it just means it doesn't find the symbols because you forgot to link to them.

Just add `MAVSDK::mission_raw` here:
https://github.com/PX4/Firmware/blob/7354e3989335dfb38cc41216182f48fe84d165c7/test/mavsdk_tests/CMakeLists.txt#L23

Also, I see you are using/extending the PX4 AutopilotTester to work for VTOL. Would you mind contributing that back so that we have some VTOL tests upstream? That would be nice!

notoriou5, 2020-07-27:
haha, such an easy fix

I will try to implement my mission and will report back to you if I hit some trouble.  If the mission ends up working fine I will upstream it :+1:

julianoes, 2020-07-27:
Cool thanks.

closed 2020-07-27 by julianoes

notoriou5, 2020-07-27:
Filling the item with stuff and trying to upload it seems to crash somewhere, but I can not narrow down the cause...

That's what I am executing:
[image]

That's the function:
[image]

And here is the error:
[image]

I played around with the information I am giving to the item, commented the push_back, skipped the mission_upload but the seg fault still persists. Any clue why?

I am just playing around a bit to understand how to make it work. For a possible upstream it would be nice to have some kind of logic behind creating missionRaw items instead of filling each one separately and pushing them back.

julianoes, 2020-07-27:
You can run the test with `--debugger gdb` which should give you a stacktrace.

> For a possible upstream it would be nice to have some kind of logic behind creating missionRaw items instead of filling each one separately and pushing them back.

That's why we have the mission plugin. So mission_raw will stay unsupported and "raw" while we're discussing on how to add VTOL capabilities to mission or creating a mission_vtol plugin. Also see the discussion in https://github.com/mavlink/MAVSDK-Proto/issues/168.

notoriou5, 2020-07-27:
> You can run the test with `--debugger gdb` which should give you a stacktrace.

When running with `--debugger gdb`, mavsdk seems to have trouble connecting with my system. At least the connect() function fails due to timeout reasons. Any clue why this is happening? I checked and gdb is installed (8.1-0ubuntu3). 

> > For a possible upstream it would be nice to have some kind of logic behind creating missionRaw items instead of filling each one separately and pushing them back.
> 
> That's why we have the mission plugin. So mission_raw will stay unsupported and "raw" while we're discussing on how to add VTOL capabilities to mission or creating a mission_vtol plugin. Also see the discussion in [mavlink/MAVSDK-Proto#168](https://github.com/mavlink/MAVSDK-Proto/issues/168).

I will check it out, thanks for pointing it out to me :+1:

julianoes, 2020-07-28:
> Any clue why this is happening?

Yer, I see gdb is actually started for px4 instead of the test, so that doesn't help.

You can run the mavsdk tests it with gdb like this:
```
diff --git a/test/mavsdk_tests/process_helper.py b/test/mavsdk_tests/process_helper.py
index 826bd6d9cb..602b862579 100644
--- a/test/mavsdk_tests/process_helper.py
+++ b/test/mavsdk_tests/process_helper.py
@@ -257,6 +257,5 @@ class TestRunner(Runner):
         self.name = "mavsdk_tests"
         self.cwd = workspace_dir
         self.env = {"PATH": os.environ['PATH']}
-        self.cmd = workspace_dir + \
-            "/build/px4_sitl_default/mavsdk_tests/mavsdk_tests"
-        self.args = ["--url", mavlink_connection, case]
+        self.cmd = "gdb"
+        self.args = ["-ex", "run", "--args", workspace_dir + "/build/px4_sitl_default/mavsdk_tests/mavsdk_tests", "--url", mavlink_connection, case]
```

notoriou5, 2020-07-28:
Thanks, starting the debugger worked :) Now I am stuck with this error message which I don't really know how to read. Could you have a look at it?

[image]

julianoes, 2020-07-28:
Ok any chance you can type `bt` enter to get a backtrace?

notoriou5, 2020-08-05:
Sorry, I didn't really work with the gdb before. But here it is :D Does this make any sense to you @julianoes ?
[image]

julianoes, 2020-08-05:
Nice!