mavlink/MAVSDK #1154: Unknown mav_frame enum value: 12 (telemetry_service_impl.h:648)

iwishiwasaneagle, 2020-07-27:
mavsdk_server show's this error `Unknown mav_frame enum value: 12 (telemetry_service_impl.h:648)` when `offboard.set_velocity_ned` or `offboard.set_velocity_body` is used on fixed-wing aircraft through MAVSDK-Python.

julianoes, 2020-07-28:
I never used or tested offboard with fixedwing. I suggest to look into the PX4 code and check what is supported.

iwishiwasaneagle, 2020-07-28:
> I never used or tested offboard with fixedwing. I suggest to look into the PX4 code and check what is supported.

Looking at the [documentation](https://docs.px4.io/master/en/flight_modes/offboard.html), it seems that velocity offboard control isn't possible for fixed-wing. However, the error is still a bit weird. If I hadn't run the `mavsdk_server` manually I most likely wouldn't have seen it.

julianoes, 2020-07-28:
It's true that this is not obvious. I guess for now we should add a note to the plugin or API description that this is for multicopter only. Pull requests welcome.

closed 2020-07-28 by iwishiwasaneagle