mavlink/MAVSDK #1157: Could we get more info about GPS?

dangviethieu, 2020-08-05:
I saw we only get 2 info: num_satellites and fix_type of [GPS in telemetry](https://mavsdk.mavlink.io/develop/en/api_reference/structmavsdk_1_1_telemetry_1_1_gps_info.html). But the qgroundcontrol is can get hdop, vdop, Course Over Ground [image](https://gyazo.com/13d2d575bc02f3fe107368a65be0c590)
So I wonder that we get more info about GPS like qgroundcontrol?
Thanks

julianoes, 2020-08-05:
Yes, that would be possible. Feel free to make a PR describing the data here: https://github.com/mavlink/MAVSDK-Proto/blob/85b5f7f270372b8395f86af50b608544a0cd455f/protos/telemetry/telemetry.proto#L422-L426

dangviethieu, 2020-08-05:
@julianoes I want to question that currently, could we get more info GPS and how to get its?

julianoes, 2020-08-05:
Currently you can't get it. You need to change the API and then we can add the functionality that you need.

dangviethieu, 2020-08-05:
@julianoes Oh. Thanks your support. I will try update the API .

julianoes, 2021-03-17:
There is no more information available, see https://github.com/mavlink/MAVSDK/pull/1333/commits/87895b22d3ee6b121ef67b117c08493182d72fda.

closed 2021-03-17 by julianoes