mavlink/MAVSDK #1162: mission_raw->mission_changed not received on mission clear all

Jonsen94, 2020-08-11:
If a mission is deleted from the drone no mission_changed event is triggered if it's deleted using the MAV_MISSION_TYPE_ALL-type. As can be seen in [0] the MAV_MISSION_TYPE type is checked. If it is MAV_MISSION_TYPE_ALL no mission_changed event will be triggered despite the mission changed (was deleted).

[0] https://github.com/mavlink/MAVSDK/blob/2e95299c656b06a46cc9594b71a169cb6095f985/src/plugins/mission_raw/mission_raw_impl.cpp#L63-L65

JonasVautherin, 2020-08-11:
Thanks for reporting! Would you be willing to open a pull request to fix this?

Jonsen94, 2020-08-11:
Sure, just give me some days.

Jonsen94, 2020-08-11:
Might be related. I Just wanted to open a bug report for the python mission_changed subscription not working. Looks like it has a related problem. QGC seems to upload missions with mission type 2 (MAV_MISSION_TYPE_RALLY) which also does not trigger a mission changed event.

Also in case of deleting a QGC mission the result type is 2. I guess this is a design choice if you want to trigger an event in any of the cases. I would suggest doing so as it is a mission in every case? At least MAVLINK calls it MissionItems even for the Geo-Fence points.
see [MAV_MISSION_TYPE](https://mavlink.io/en/messages/common.html#MAV_MISSION_TYPE)

Still gonna open the python-report and link it so others know about it.

Jonsen94, 2020-08-12:
@JonasVautherin : it seems like the problem is even deeper.

As far as I can tell then drone just doesn't sent the MISSION_ACK command to anyone except for QGC.
I even tried running QGC with the same sys-id as mavsdk. QGC reports MISSION_ACK messages with sysid 1 and comp id 190 but the mavsdk does not receive such a packet. It seems like this message alone is not enough to report mission changes, not sure how QGC does it so accurate.

JonasVautherin, 2020-08-12:
Can you describe your setup? Is it something like below?

* PX4 running on the drone
* QGC running on the laptop, on udp://:14550
* MAVSDK running on the laptop, on udp://:14540

Can you reproduce the bug with SITL?

Jonsen94, 2020-08-12:
I can't test it with a real drone ;)

Its the setup described in https://github.com/mavlink/MAVSDK-Python/issues/244 except for the python part.

Drone is a px4-gazebo setup from https://github.com/PX4/Firmware.git, I am using tag v1.10.2

julianoes, 2020-08-24:
> QGC seems to upload missions with mission type 2 (MAV_MISSION_TYPE_RALLY) which also does not trigger a mission changed event.

This would be expected, because rally points are not technically mission items. And they are not supported with MAVSDK yet, I think.

Jonsen94, 2020-08-25:
Feels a bit strange I would say. I mean, it is called "MAV_MISSION_[...]". 
Does that mean that the current behavior is the intended one?

If so this should probably be noted somewhere in the documentation. I mean the API endpoint states that it is meant to be used to detect mission uploads from other GC stations and QGC is as far as I know one of the most widely used ones ? 

Or could we just add another parameter (mission_type) into the detection which is returned so one knows which kind of "mission" was changed?

julianoes, 2020-08-25:
It is called MAV_MISSION_ because it re-uses the mission messages for upload. But the "MISSION_TYPE" is RALLY.

I would say it's the correct implementation but we should add a `rally_points_changed()` notification for rally points.

julianoes, 2020-11-23:
I'm closing this because I think this would need to be fixed as part of a rally points plugin. To me it has not much to do with the mission_raw plugin which only handles MAV_MISSION_TYPE_MISSION for now.

@Jonsen94 feel free to start a PR in https://github.com/mavlink/MAVSDK-Proto if you want to add a rally plugin.

closed 2020-11-23 by julianoes