mavlink/MAVSDK #1164: GMock Linker Error: Build MAVSDK On Windows

tpool2, 2020-08-12:
I've had trouble building MAVSDK on Windows from source following documentation found [here](https://mavsdk.mavlink.io/develop/en/contributing/build.html). After installing CMake, and Visual Studio 2019 C++ Build Tools, my steps are as follows in Powershell

1. git clone --depth=50 --branch=master https://github.com/mavlink/MAVSDK.git mavlink/MAVSDK
2. cd .\mavlink\MAVSDK\
3. git checkout master
4. git submodule update --init --recursive

5. cmake -G "Visual Studio 16 2019" -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=install -Bbuild/default -S.

This step works fine, but when I proceed to 

6. cmake --build build/default --config Release --target install

I get a linker error about gmock. (Output copied/pasted below)

mavlink_mission_transfer_test.obj : error LNK2019: unresolved external symbol "class testing::internal::Mutex testing::internal::g_gmock_mutex" (?g_gmock_mutex@internal@testing@@3VMutex@12@A) referenced in function "private:
virtual void __cdecl MAVLinkMissionTransfer_UploadMissionSendsCount_Test::TestBody(void)" (?TestBody@MAVLinkMissionTransfer_UploadMissionSendsCount_Test@@EEAAXXZ) [D:\Users\pool6\mavlink\MAVSDK\build\default\src\unit_tests_ru
nner.vcxproj]
mavlink_mission_transfer_test.obj : error LNK2019: unresolved external symbol "class testing::internal::ThreadLocal<class testing::Sequence *> testing::internal::g_gmock_implicit_sequence" (?g_gmock_implicit_sequence@internal
@testing@@3V?$ThreadLocal@PEAVSequence@testing@@@12@A) referenced in function "protected: class testing::internal::TypedExpectation<bool __cdecl(struct __mavlink_message &)> & __cdecl testing::internal::FunctionMocker<bool __
cdecl(struct __mavlink_message &)>::AddNewExpectation(char const *,int,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class std::tuple<class testing::Matcher<struct __mavlink_m
essage &> > const &)" (?AddNewExpectation@?$FunctionMocker@$$A6A_NAEAU__mavlink_message@@@Z@internal@testing@@IEAAAEAV?$TypedExpectation@$$A6A_NAEAU__mavlink_message@@@Z@23@PEBDHAEBV?$basic_string@DU?$char_traits@D@std@@V?$al
locator@D@2@@std@@AEBV?$tuple@V?$Matcher@AEAU__mavlink_message@@@testing@@@6@@Z) [D:\Users\pool6\mavlink\MAVSDK\build\default\src\unit_tests_runner.vcxproj]
D:\Users\pool6\mavlink\MAVSDK\build\default\src\Release\unit_tests_runner.exe : fatal error LNK1120: 2 unresolved externals [D:\Users\pool6\mavlink\MAVSDK\build\default\src\unit_tests_runner.vcxproj]
PS D:\Users\pool6\mavlink\MAVSDK> ls .\build\

Any suggestions would be much appreciated. When I get this working, I would be more than happy to open a pull request for updating the documentation.

tpool2, 2020-08-14:
Update: I got it to compile by setting -DBUILD_SHARED_LIBS=OFF. It seems that setting -DBUILD_SHARED_LIBS=ON messes up GTest Linking currently. Some people [here](https://stackoverflow.com/questions/46025439/unresolved-external-symbol-error-with-google-mock-and-vcpkg) have suggested setting a preprocessor definition GTEST_LINKED_AS_SHARED_LIBRARY to the project when doing so... I'm not sure how clean that would be though. Anyways, thanks!

JonasVautherin, 2020-08-17:
Thanks for reporting! Feel free to open a PR if you later find a way to fix it. In the meantime, I'll close this one :slightly_smiling_face:.

closed 2020-08-17 by JonasVautherin

jonyMarino, 2020-12-31:
I faced the same issue. @JonasVautherin, won't it be a good idea to change the docs in the meantime?

JonasVautherin, 2021-01-04:
I am not working on Windows, so all I know is that it works on _my Windows_ (may not be the usual setup since I installed it just for MAVSDK), and it seems like it works [in the CI](https://github.com/mavlink/MAVSDK/blob/develop/.github/workflows/main.yml#L477-L510).

Happy to review any PR if you have suggestions to improve the Windows workflow :blush:.

smotrjashka, 2021-07-18:
just for motice, I have the same issue

julianoes, 2021-07-20:
@smotrjashka does https://github.com/mavlink/MAVSDK/issues/1164#issuecomment-674197395 help?