mavlink/MAVSDK #1171: Euler Angles Wrong?

jtroeth1, 2020-08-21:
Hi there,

I am subscribing to euler angle information such as:

    telemetry->subscribe_attitude_euler( [&](Telemetry::EulerAngle euler) {
                setTimestamp(stateMachine->data.uavPose_.timestamp);                       
                stateMachine->data.uavPose_.roll = euler.roll_deg;
                stateMachine->data.uavPose_.pitch = euler.pitch_deg;
                stateMachine->data.uavPose_.yaw = euler.yaw_deg;
                std::cout << "Euler: (" << euler.roll_deg << ", " << euler.pitch_deg << ", " << euler.yaw_deg << ")" << std::endl;
     }); 

It seems every 6th return is a vastly different result from most:

Euler: (-0.098097, 0.0927901, -0.586002)
Euler: (-0.0981, 0.0927703, -0.585981)
Euler: (-0.0980991, 0.0927632, -0.585978)
Euler: (-0.0981108, 0.0927819, -0.585949)
Euler: (-0.0981177, 0.0927954, -0.585942)
**Euler: (114.939, 41.1869, -27.6619)**
Euler: (-0.0981175, 0.0927945, -0.585928)
Euler: (-0.0981172, 0.0927943, -0.585894)
Euler: (-0.0981132, 0.092764, -0.585896)
Euler: (-0.0981357, 0.0927864, -0.585868)
Euler: (-0.0981253, 0.0927955, -0.585837)
**Euler: (114.94, 41.1802, -27.6554)**
Euler: (-0.0981328, 0.0928163, -0.585833)
Euler: (-0.0981107, 0.0928, -0.585827)
Euler: (-0.0980776, 0.0927828, -0.585837)
Euler: (-0.0981029, 0.092776, -0.585825)
Euler: (-0.0981031, 0.0927992, -0.585849)
**Euler: (114.94, 41.1809, -27.6564)**

Do you have any idea what the reason for this is?
I am connected via serial at rate 57600. Callback is at 10Hz.

Thanks.

JonasVautherin, 2020-08-21:
I may have observed something similar with the attitude quaternion. Would you mind checking if every 6th quaternion is also wrong?

jtroeth1, 2020-08-23:
Hi @JonasVautherin ,

Yes same thing:
Quaternion: (0.999855, 0.000940929, -0.001161, -0.0169413)
Quaternion: (0.999855, 0.000940839, -0.00116119, -0.0169412)
Quaternion: (0.999855, 0.000940866, -0.00116114, -0.0169413)
Quaternion: (0.999855, 0.000940803, -0.00116107, -0.016941)
Quaternion: (0.999855, 0.000940805, -0.00116105, -0.0169412)
Quaternion: (-0.565573, -0.565573, 0.0822511, 0.820263)
Quaternion: (0.999855, 0.000940897, -0.00116118, -0.0169412)
Quaternion: (0.999855, 0.000940888, -0.00116108, -0.0169414)
Quaternion: (0.999855, 0.000940994, -0.0011609, -0.0169415)
Quaternion: (0.999855, 0.000940872, -0.00116085, -0.0169415)
Quaternion: (0.999855, 0.000940928, -0.00116081, -0.0169416)
Quaternion: (-0.565555, -0.565555, 0.0822484, 0.820276)

jtroeth1, 2020-08-23:
I think it is a callback issue for eg. if I do:

    while(true){
        Telemetry::Quaternion quaternion = telemetry->attitude_quaternion();
        std::cout << "Quaternion: (" << quaternion.w << ", " << quaternion.x << ", " << quaternion.y << ", " << quaternion.z << ")" << std::endl;
        sleep_for(milliseconds(100));
    }

There is no issue, the return value's are correct...

julianoes, 2020-08-24:
@jtroeth1 which version are you using? This should have been fixed in https://github.com/mavlink/MAVSDK/pull/1155.

jtroeth1, 2020-08-24:
Hi @julianoes,
I am on:
v0.28.0 / a1e83140
Thanks

julianoes, 2020-08-24:
Sorry, I want to be sure. Can you post the full output of the program that you run?

jtroeth1, 2020-08-25:
I'm not sure what you mean sorry? The full output? That is the full output.

For the purposes of this question the code is:

    #include <chrono>
    #include <cstdint>
    #include <mavsdk/mavsdk.h>
    #include <mavsdk/plugins/action/action.h>
    #include <mavsdk/plugins/telemetry/telemetry.h>
    #include <iostream>
    #include <thread>

    using namespace mavsdk;
    using namespace std::this_thread;
    using namespace std::chrono;

    #define ERROR_CONSOLE_TEXT "\033[31m" // Turn text on console red
    #define TELEMETRY_CONSOLE_TEXT "\033[34m" // Turn text on console blue
    #define NORMAL_CONSOLE_TEXT "\033[0m" // Restore normal console colour

    void usage(std::string bin_name)
    {
        std::cout << NORMAL_CONSOLE_TEXT << "Usage : " << bin_name << " <connection_url>" << std::endl
                  << "Connection URL format should be :" << std::endl
                  << " For TCP : tcp://[server_host][:server_port]" << std::endl
                  << " For UDP : udp://[bind_host][:bind_port]" << std::endl
                  << " For Serial : serial:///path/to/serial/dev[:baudrate]" << std::endl
                  << "For example, to connect to the simulator use URL: udp://:14540" << std::endl;
    }

    void component_discovered(ComponentType component_type)
    {
        std::cout << NORMAL_CONSOLE_TEXT << "Discovered a component with type "
                  << unsigned(component_type) << std::endl;
    }

    int main(int argc, char** argv)
    {
        Mavsdk dc;
        std::string connection_url;
        ConnectionResult connection_result;

        bool discovered_system = false;
        if (argc == 2) {
            connection_url = argv[1];
            connection_result = dc.add_any_connection(connection_url);
        } else {
            usage(argv[0]);
            return 1;
        }

        if (connection_result != ConnectionResult::Success) {
            std::cout << ERROR_CONSOLE_TEXT << "Connection failed: " << connection_result
                      << NORMAL_CONSOLE_TEXT << std::endl;
            return 1;
        }

        // We don't need to specify the UUID if it's only one system anyway.
        // If there were multiple, we could specify it with:
        // dc.system(uint64_t uuid);
        System& system = dc.system();

        std::cout << "Waiting to discover system..." << std::endl;
        dc.register_on_discover([&discovered_system](uint64_t uuid) {
            std::cout << "Discovered system with UUID: " << uuid << std::endl;
            discovered_system = true;
        });

        // We usually receive heartbeats at 1Hz, therefore we should find a system after around 2
        // seconds.
        sleep_for(seconds(2));

        if (!discovered_system) {
            std::cout << ERROR_CONSOLE_TEXT << "No system found, exiting." << NORMAL_CONSOLE_TEXT
                      << std::endl;
            return 1;
        }

        // Register a callback so we get told when components (camera, gimbal) etc
        // are found.
        system.register_component_discovered_callback(component_discovered);

        auto telemetry = std::make_shared<Telemetry>(system);
        auto action = std::make_shared<Action>(system);

        // We want to listen to the altitude of the drone at 1 Hz.
        const Telemetry::Result set_rate_result = telemetry->set_rate_position(1.0);
        if (set_rate_result != Telemetry::Result::Success) {
            std::cout << ERROR_CONSOLE_TEXT << "Setting rate failed:" << set_rate_result
                      << NORMAL_CONSOLE_TEXT << std::endl;
            return 1;
        }

        telemetry->subscribe_attitude_euler( [&](Telemetry::EulerAngle euler) {
               std::cout << "Euler: (" << euler.roll_deg << ", " << euler.pitch_deg << ", " << euler.yaw_deg << ")" << std::endl;
        }); 

        //telemetry->subscribe_attitude_quaternion( [&](Telemetry::Quaternion quaternion) {
            //std::cout << "Quaternion: (" << quaternion.w << ", " << quaternion.x << ", " << quaternion.y << ", " << quaternion.z << ")" << std::endl;
        //});

        // We are relying on auto-disarming but let's keep watching the telemetry for a bit longer.
        //sleep_for(seconds(3));

        while(true){
            //Telemetry::Quaternion quaternion = telemetry->attitude_quaternion();
            //std::cout << "Quaternion: (" << quaternion.w << ", " << quaternion.x << ", " << quaternion.y << ", " << quaternion.z << ")" << std::endl;
            //sleep_for(milliseconds(100));
        }
        std::cout << "Finished..." << std::endl;

        return 0;
    }

This produces the output in the initial question.
Thanks

julianoes, 2020-08-25:
No, sorry, I meant the full log output.

jtroeth1, 2020-09-21:
Sorry, I'm not sure what logs you mean?

> I think it is a callback issue for eg. if I do:
>
> ```
> while(true){
>     Telemetry::Quaternion quaternion = telemetry->attitude_quaternion();
>     std::cout << "Quaternion: (" << quaternion.w << ", " << [...]

Also In the above, it is actually all data is incorrect outside of the callback.

julianoes, 2020-09-22:
> I am on:
v0.28.0 / 6d01d43 develop
Thanks

It has been fixed since v0.29.0, please update. I'm closing this, re-open if you find it again.

closed 2020-09-22 by julianoes