mavlink/MAVSDK #1173: Error message if no manual control is available

julianoes, 2020-08-24:
It would be nice to have an error message if starting altitude or position control using manual_control fails.

This would be similar as it is done for offboard: https://github.com/mavlink/MAVSDK/blob/758a6437826aa1e6a1abda5f109a7bca21599d61/src/plugins/offboard/offboard_impl.cpp#L43

yashom7, 2020-10-07:
Hi!
I think I can get on with implementing this.

Just to clarify. This error message reporting will go in the Manual Control plugin and is used if some means of control i.e. a joystick is not available?

afwilkin, 2020-10-21:
From what I am understanding (and can confirm by testing), manual control needs a setpoint before entering the mode. This is an example for offboard `offboard->set_velocity_body({0,` 0, 0, 0}); `
So, in code it looks like this:
`auto offboard = std::make_shared<Offboard>(*system);
 offboard->set_velocity_body({0, 0, 0, 0}); /* Needed */
 Offboard::Result offboard_result = offboard->start();`

Without the `offboard->set_velocity_body({0, 0, 0, 0}); ` then we get the Result shown in Julian's comment above (Offboard::Result::NoSetpointSet;). 

**Manual Control**
Manual Control needs to be started in a very similar fashion

`manual_control->set_manual_control_input(0.f, 0.f, 0.f, 0.f); /*Needed*/
 auto manual_control_result = manual_control->start_position_control();`

However, if the `manual_control->set_manual_control_input(0.f, 0.f, 0.f, 0.f);` is NOT there, manual control will not be achieved. This should be an error. However, the code returns ManualControl::Result::Success instead of a result that would indicate that manual control is not going to work. (For example: ManualControl::Result::NoSetpointSet)

**TODO**
Implement something like ManualControl::Result::NoSetpointSet that will tell the developer they need to insert `manual_control->set_manual_control_input(0.f, 0.f, 0.f, 0.f); ` into their code

@julianoes is that a fair analysis?

julianoes, 2020-10-22:
Sorry for the late response here.

> Just to clarify. This error message reporting will go in the Manual Control plugin and is used if some means of control i.e. a joystick is not available?

Yes.

> @julianoes is that a fair analysis?

100%, thanks for writing that up @afwilkin.

PR #1261 (2020-11-20): manual_control: use proper result if input not set
A result/error of unknown is not a good user feedback. Instead we need
to communicate the actual reason.

Closes #1173.
closed 2020-11-20 by julianoes