mavlink/MAVSDK #1182: MAVSDK failed to initiate UDP connection

cjffly, 2020-08-28:
I have strange issue when I try to use MAVSDK C++ application to start a connection using UDP:
I got a "pixhawk - raspberry pi(as a bridge) - wifi router - ubuntu machine" setup. While I am trying to run some MAVSDK application, such as take_off_land example like `./take_off_land udp://:14540`, it will tell me `No system found, exiting.`

When I open QGC on the ubuntu machine, it still won't automatically find any system. However, when I manually add udp connection throught QGC, specify the IP, like `192.168.0.101:14540`, QGC will find the system and connect to my pixhawk. 

Even strangely, I close QGC at this moment, and run the MAVSDK application `./take_off_land udp://:14540` on the same ubuntu machine, it recognize the system and let the drone take off.

Can someone tell me is this a bug in MAVSDK? or the QGC manual UDP connection is different than udp connection used by MAVSDK?

JonasVautherin, 2020-08-28:
I think it's a networking issue. When you run `./take_off_land udp://:14540`, what MAVSDK does is that it waits until it receives HEARTBEAT messages from the drone over UDP. QGC does the same. Apparently, the network broadcasts coming from your drone don't reach the Ubuntu machine. That's not unsurprising, many network topology behave the same.

By specifying the RPi's IP (`192.168.0.101`) to QGC, my guess is that QGC started sending heartbeats to it, which "opened" the connection. And then the RPi started talking to your Ubuntu machine instead of broadcasting, which is why MAVSDK got the messages. MAVSDK does not initiate the connection.

Can you modify the RPi side so that it sends MAVLink directly to your Ubuntu IP address, to verify that?

cjffly, 2020-08-28:
Thanks for the advise. I will try to do something on the RPi.

Is it possible using MAVSDK to send a heartbeats just like what QGC did?

JonasVautherin, 2020-08-28:
The mavftp server example does it: https://github.com/mavlink/MAVSDK/blob/develop/examples/ftp_server/ftp_server.cpp#L43

I wonder if you could do that [here](https://github.com/mavlink/MAVSDK/blob/develop/examples/takeoff_land/takeoff_and_land.cpp#L47):

```c++
connection_result = mavsdk.setup_udp_remote(...);
```

instead of 

```c++
connection_result = dc.add_any_connection(...);
```

If you try it, let me know what happens :slightly_smiling_face:

cjffly, 2020-08-29:
```c++
> connection_result = mavsdk.setup_udp_remote(...);
> ```

That actually did work! Thanks!

The only difference is if I use `add_any_connection(...)` , the argv would be something like udp://:14540, using `setup_udp_remote(...) requires the ip and port as separate arguments. 

Correct me if I am wrong: Once the connection is setup, this udp remote method handle the communication the same as add_any_connection method, right?

cjffly, 2020-08-29:
Another problem is the remote method seems only work with one system at a time. When dealing with multiple systems, like [here](https://github.com/mavlink/MAVSDK/blob/3992a492b28fac8687213d7f81fcff0420b0e9b5/examples/multiple_drones/multiple_drones.cpp#L41)
```
        ConnectionResult connection_result = dc.add_any_connection(argv[i]);
```
Replaced with
```
        ConnectionResult connection_result = dc.setup_udp_remote(argv[i],argv[i+1]);
```

The system seems only find the last argument, not all systems.

JonasVautherin, 2020-08-29:
> The system seems only find the last argument, not all systems.

Hmm, it could be that it only has one udp remote, at the moment. Not sure how it could be edited to add multiple ones. Feel free to have a look and open a PR if you find a solution :blush:.

I'm closing this in the meantime, let's open a new issue if you want to discuss the multiple `setup_udp_remote`.

closed 2020-08-29 by JonasVautherin