mavlink/MAVSDK #1183: Hold time is ignored for waypoints when importing with import_qgroundcontrol_mission

petergerten, 2020-09-01:
On waypoint import the hold time for a waypoint is ignored.
See https://github.com/mavlink/MAVSDK/blob/37be38dad47d575f29d76f886e7539e4f419ed67/src/plugins/mission/mission_impl.cpp `Mission::Result MissionImpl::build_mission_items`

All current code to parse a waypoint:
```
            if (command == MAV_CMD_NAV_WAYPOINT) {
                auto is_fly_through = !(int(params[0]) > 0);
                new_mission_item.is_fly_through = is_fly_through;
            }
            auto lat = params[4], lon = params[5];
            new_mission_item.latitude_deg = lat;
            new_mission_item.longitude_deg = lon;

            auto rel_alt = float(params[6]);
            new_mission_item.relative_altitude_m = rel_alt;
```
The hold time in params[0] is already used to determine if it is a 'fly-though' waypoint.
But then it is not used to set the hold time in seconds.

JonasVautherin, 2020-09-01:
Nice catch! Would you mind opening a PR to fix that?

petergerten, 2020-09-02:
see https://github.com/mavlink/MAVSDK/pull/1184

PR #1184 (2020-09-07): Fixed loiter time (hold time) being ignored on QGC mission import.
Resolves #1183.
closed 2020-09-07 by JonasVautherin