mavlink/MAVSDK #1186: error messages on actions do not reveil error reason

petergerten, 2020-09-06:
For example if I arm:
```
try:
  await drone.action.arm()
except mavsdk.action.ActionError as err:
  print(err)
```
The error result is:
`COMMAND_DENIED: 'Command Denied'; origin: arm(); params: () `

But I don't know what caused the error. The PX4 SITL shell in this case shows:

`WARN  [commander] Preflight Fail: High Accelerometer Bias`

This seems to also propagate via MAVLink as I can see it in QGC.
But it seems there is no way at the moment to get this message via MAVSDK - or am I missing something?

julianoes, 2020-09-07:
Good point. The problem is that in MAVLink these "error messages" are only statustext information but the ack itself does not tell the reason.

You could subscribe to statustext in Python (https://github.com/mavlink/MAVSDK-Proto/blob/8b7dc4e1e8f34de9a8dbf9f66e1e832104f26d21/protos/telemetry/telemetry.proto#L47-L48) to get that information but that's awkward.

Alternatively, we could enable these printfs to happen in Python no matter if you subscribed or not but it's a bit ugly too.

The optimal real solution would be an interface like `is_armable` that gives you a detailed report of what is green and what is still red in order to arm. There is the MAVLink message [SYS_STATUS](https://mavlink.io/en/messages/common.html#SYS_STATUS) for it and we should try more and more rely on it (instead of statustext messages), so more work is needed on the PX4 side but also MAVSDK side. And even more long term there is an [events interface](https://docs.google.com/document/d/18qdDgfML97lItom09MJhngYnFzAm1zFdmlCKG7TaBpg/edit) planned which would address that even better.

petergerten, 2020-09-08:
Thanks - yes, I missed the StatusText in Telemetry - it seems to be the easiest option at the moment. While it is useful for human use, it is not ideal for machine parsing though. I will look into SYS_STATUS. Also the planned events interface looks very interesting.

closed 2024-05-27 by julianoes