mavlink/MAVSDK #1198: Reading GPS RAW data other than fused one

cjffly, 2020-09-27:
Hi,

I am trying to use MAVSDK to read a GPS height data from px4 (RTK connected), right now I am subscribing to telemetry-position: 
```
telemetry->subscribe_position([](Telemetry::Position position) {
z = position.absolute_altitude_m;
```
However, I did notice the height from the telemetry->position is fused and not accurate compare the raw RTK-GPS height. (I already set the `EKF2_HGT_MODE = GPS` ), the flight log show sometimes the fused height estimation is drifting around even my system is sitting on the ground:
[image]

My question is, is there a way for MAVSDK to read the mavlink message GPS_RAW_INT, other than the fused data from (telemetry->position) showed in the flight log above?

Best
C.J.

julianoes, 2020-09-28:
Ok, that's an interesting observation. Now the question is what exactly you need from MAVSDK. Do you need the GPS height for debugging? I'm asking because to me it seems that this is more of a PX4 estimation and controls problem than a MAVSDK problem.

cjffly, 2020-09-28:
I am using that GPS data as a reference point, which my other applications will rely on. I did get some answer from px4 perspective [here](https://discuss.px4.io/t/difference-between-gps-altitude-vs-fused-altiude-estimation/18784), seems the fused data also consider other sensors. 

So the question becomes is there any method for MAVSDK to access the raw data, not fused ones?

Also, I notice in MAVLink message definition, there are [GPS_RAW_INT](https://mavlink.io/en/messages/common.html#GPS_RAW_INT) and [GLOBAL_POSITION_INT](https://mavlink.io/en/messages/common.html#GLOBAL_POSITION_INT). Does MAVSDK provide any api to access that?

julianoes, 2020-10-05:
@cjffly if you are using MAVSDK C++ you can use the Passthrough plugin to access any MAVLink message. So that would be one way. Alternatively, you can make a pull request to https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/telemetry/telemetry.proto to extend the Telemetry plugin to include an API for Raw GPS.

julianoes, 2021-03-26:
This has been added, see https://github.com/mavlink/MAVSDK/pull/1333.

closed 2021-03-26 by julianoes