mavlink/MAVSDK #1207: process_fixedwing_metrics does not return groundspeed, alt and heading

haydockjp, 2020-10-06:
mavlink_msg_vfr_hud_decode gives us access to additional data `groundspeed, alt and heading` but process_fixedwing_metrics does not return them

haydockjp, 2020-10-06:
https://github.com/mavlink/MAVSDK/pull/1208

JonasVautherin, 2020-10-07:
One question I have here: don't those values already come from other APIs? E.g. Altitude comes from `subscribe_position`, etc. Or are those different?

haydockjp, 2020-10-07:
I mainly want to get ground speed. I thought I would add the others as they are on the original mavlink message. Altitude is definately somewhere else. I'm not sure about heading. I haven't seen it do far.

I'm also trying to use this SDK against ardipilot. I know you only officially support PX4, but it's mostly working for me.

JonasVautherin, 2020-10-07:
I would have thought that:

* Bearing comes from [attitude](https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/telemetry/telemetry.proto#L25-L28)
* Altitude comes from [position](https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/telemetry/telemetry.proto#L15-L16)
* Ground speed comes from [velocity](https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/telemetry/telemetry.proto#L35-L36)

The idea being that ideally, we would want only one interface to access one value, otherwise I am concerned we may end up on confusing situations... So I am interested in knowing the use-case for adding those values to FixedWingMetrics (maybe there is a need).

> I'm also trying to use this SDK against ardipilot. I know you only officially support PX4, but it's mostly working for me.

Oh really? That's cool! Would you mind reporting on https://github.com/mavlink/MAVSDK/issues/728? We support PX4 because we don't work with ArduPilot, but we would love to take contributions! If you're already using MAVSDK with ArduPilot, it would be amazing if you could explain what you do on #728. Like issues you encounter, features you use, features that don't work, etc.

MAVSDK could detect if you are running ArduPilot and change the behavior when needed. We just need contributors using ArduPilot to tell us what's needed 😊.

haydockjp, 2020-10-07:
We currently use https://ardupilot.org/planner/ to run missions. I'm looking to get the same telemetry into our own data store. It gets those fields from the Fixedwing Metrics telemetry.

If you are looking for differences between PC4 and Ardupilot, Mission Planner is open source and may be a good resource.

I'll have a look at the other telemetry that you suggested.

JonasVautherin, 2020-10-07:
> If you are looking for differences between PX4 and Ardupilot, Mission Planner is open source and may be a good resource.

I'm actually looking for contributions from ArduPilot users :innocent:. MAVSDK being developed against PX4 and not ArduPilot is a question of resources, mainly. We would love to accept contributions from anybody interested in doing something for ArduPilot.

As you mentioned that MAVSDK "mostly works for you [on ArduPilot]", I was hoping to get insights like "the telemetry plugin works, those actions work, and uploading a mission doesn't", or whatever you know from your experience.

> I'll have a look at the other telemetry that you suggested.

Sure, keep us posted! If those don't give you the values you need, we can add them to `FixedWingMetrics`. But if they work for you, I would rather not duplicate them in the API :+1:.

haydockjp, 2020-10-07:
I might be misunderstanding some of these values, but:

*Altitude:* position - I am getting this already. I actually don't really need this as I am controlling a boat, but altitude might be interesting to see if it could indicate wave heave (same with climb rate)

*Ground Speed:* velocity 
=> `SubscribeVelocityNed` this gives the velocity along three accesses. I just want the ground speed

*Bearing:* attitude =>I was just interested in the compass bearing
 => `EulerAngle` - this gives roll, pitch and yaw
 => `Quaternion` - this gives reference in w, x, y and z -

JonasVautherin, 2020-10-07:
> => SubscribeVelocityNed this gives the velocity along three accesses. I just want the ground speed

Can't you compute the ground speed based on the NED velocity vector? :thinking: 

> Bearing: attitude =>I was just interested in the compass bearing

Same question here...

haydockjp, 2020-10-07:
> I'm actually looking for contributions from ArduPilot users 😇. MAVSDK

On the mission side. I am now able to start, pause and clear using Mission.
I can download using Mission Raw (Not Mission). I do have an issue with
uploading a mission. Using Mission Raw I get a time out. I still need to
look into why

julianoes, 2020-10-22:
> Using Mission Raw I get a time out. I still need to look into why

I would assume you have something in there that's not understood on either ArduPilot or MAVSDK's side.

> Can't you compute the ground speed based on the NED velocity vector? :thinking: 

Right, I think:
```
float bearing_deg = tan(speed_x/speed_y) * 180.0f / M_PI;
float ground_speed_m_s = sqrt(speed_x*speed_x + speed_y*speed_y);
```

@JonasVautherin I think it would make sense to add these two fields to the API for velocity NED, so that they are easily accessible.

closed 2020-11-20 by julianoes