mavlink/MAVSDK #1213: Offboard disabled when processing other components heartbeat

yhabib29, 2020-10-08:
When using Offboard plugin, I have noticed that the plugin is processing heartbeats coming from other components on the same system (Camera, Gimbal, ...).
My workaround is to filter Heartbeats mavlink message based on the message component ID, using default Autopilot value MAV_COMP_ID_AUTOPILOT1.

I just added these lines at the beginning of the "process_heartbeat" function:
```
void OffboardImpl::process_heartbeat(const mavlink_message_t& message)
{
    // Process only Heartbeat coming from the autopilot
    if (message.compid != MAV_COMP_ID_AUTOPILOT1) {
        return;
    }
```

JonasVautherin, 2020-10-08:
I believe it makes sense. @julianoes: Is that correct?

julianoes, 2020-10-22:
Good point! We would appreciate a PR with the fix.

yhabib29, 2020-10-23:
No problem, I will try to do it during this week-end.

closed 2020-10-27 by yhabib29