mavlink/MAVSDK #1253: where can i find the mavsdk_server file?

kois86, 2020-11-13:
so that i can perform multiple drones setup on sitl? can't find it in the Mavsdk-python folder.

thank you in advance

JonasVautherin, 2020-11-13:
First, you can get it from the [release artifacts of the mavsdk repo](https://github.com/mavlink/mavsdk/releases) (note: this is not mavsdk-python, but mavsdk-python uses mavsdk).

Alternately, if you install mavsdk-python with `pip3 install mavsdk`, `mavsdk_server` comes embedded with it. For instance, say I install it in a venv:

```
mkdir /tmp/test_venv && cd /tmp/test_venv
python3 -m venv venv
source ./venv/bin/activate
pip install mavsdk
find . | grep mavsdk_server
```

I will get the following output:

```
% find . | grep mavsdk_server
./venv/lib/python3.8/site-packages/mavsdk/bin/mavsdk_server
```

kois86, 2020-11-14:
> First, you can get it from the [release artifacts of the mavsdk repo](https://github.com/mavlink/mavsdk/releases) (note: this is not mavsdk-python, but mavsdk-python uses mavsdk).
>
> Alternately, if [...]

thanks. 

when i do  this >>find . | grep mavsdk_server

the output is as below. is this correct? if so, i need to navigate to this directory

./.local/lib/python3.6/site-packages/mavsdk/bin/mavsdk_server

to perform this command?
./mavsdk_server -p 50040

thanks again

./MAVSDK/templates/mavsdk_server/call.j2
./MAVSDK/templates/mavsdk_server/file.j2
./MAVSDK/templates/mavsdk_server/enum.j2
./MAVSDK/templates/mavsdk_server/stream.j2
./MAVSDK/templates/mavsdk_server/struct.j2
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir/grpc_server.cpp.o
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir/build.make
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir/cmake_clean.cmake
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir/link.txt
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir/backend_api.cpp.o
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir/CXX.includecache
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir/progress.make
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir/depend.make
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir/DependInfo.cmake
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir/depend.internal
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir/backend.cpp.o
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server.dir/flags.make
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server_bin.dir
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server_bin.dir/build.make
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server_bin.dir/cmake_clean.cmake
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server_bin.dir/link.txt
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server_bin.dir/CXX.includecache
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server_bin.dir/progress.make
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server_bin.dir/depend.make
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server_bin.dir/DependInfo.cmake
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server_bin.dir/depend.internal
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server_bin.dir/mavsdk_server.cpp.o
./MAVSDK/build/default/src/backend/src/CMakeFiles/mavsdk_server_bin.dir/flags.make
./MAVSDK/build/default/src/backend/src/mavsdk_server
./MAVSDK/build/default/src/backend/src/libmavsdk_server.so
./MAVSDK/src/backend/src/mavsdk_server.cpp
./.local/lib/python3.6/site-packages/mavsdk/bin/mavsdk_server

JonasVautherin, 2020-11-14:
Yes, that's correct :+1:

kois86, 2020-11-14:
Yes, thank you Jonas, i manage to run the mavsdk_server. thank you very much.

though i received the error below when i run. i think i found the same error somewhere in the forum. will look onto it. thanks

[03:55:28|Error] bind error: Address in use (udp_connection.cpp:85)
[03:55:28|Error] Connection failed: Bind error (connection_initiator.h:41)
^C

JonasVautherin, 2020-11-14:
You need to tell mavsdk-python to not start `mavsdk_server` itself (because you start it manually). That's when you create the `System`:

```
from mavsdk import System

drone = System(mavsdk_server_address="localhost", port=50051)
```

Note how the port is what you passed to `./mavsdk_server -p 50051`. You'll need multiple ports if you start multiple instances of mavsdk_server, one for each drone.

closed 2020-11-14 by JonasVautherin

kois86, 2020-11-16:
> You need to tell mavsdk-python to not start `mavsdk_server` itself (because you start it manually). That's when you create the `System`:
>
> ```
> from mavsdk import System
>
> drone = [...]

i do this -->  drone = System(mavsdk_server_address="localhost", port=50051)

in order to assign drone to the system. for example, for multiple drone i would write like this:

 drone = System(mavsdk_server_address="localhost", port=50040)
    await drone.connect()
    drone1 = System(mavsdk_server_address="localhost", port=50041)
    await drone1.connect()

if i simply write like this --->  drone = System()

how can i assign for drone1. doesn't drone1=System() will contradict the above? sorry for the noob question, or do i have to open another issue?

thanks again

JonasVautherin, 2020-11-16:
> if i simply write like this ---> drone = System()

No you can't. That's why you need to write the other one you pasted above.

kois86, 2020-11-18:
Hi Jonas,

I have question, does this excerpt important?

#This waits till a mavlink based drone is connected
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone with UUID: {state.uuid}")
            break

because with this, when i run my mavsdk_server manually, the drone wont connect, but without it will, but some of the APIs wont work

thank you

JonasVautherin, 2020-11-18:
Which version of `mavsdk_server` are you running?

kois86, 2020-11-19:
> Which version of `mavsdk_server` are you running?

Sorry, im not sure which version and how to check it.

thing is, my program will stuck after it is successfully connect to the mavsdk_server due to the excerpt i mentioned above. if i remove it will run the program, such as takeoff and goto location, but some API cant work for example follow_me.

Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!
Waiting for mavsdk_server to be ready...
Connected to mavsdk_server!

below is the program that i plan to run. Noted that im using 2 iris drone. TQVM

`#!/usr/bin/env python3
import asyncio
from mavsdk import System
from mavsdk.follow_me import (Config, FollowMeError, TargetLocation)

default_height = 8.0 #in Meters
follow_distance = 2.0 #in Meters, this is the distance that the drone will remain away from Target while following it
#Direction relative to the Target
#Options are NONE, FRONT, FRONT_LEFT, FRONT_RIGHT, BEHIND
direction = Config.FollowDirection.BEHIND
responsiveness =  0.02

#./mavsdk_server -p 50040 udp://:14540
#./mavsdk_server -p 50041 udp://:14541

async def run():
    drone = System(mavsdk_server_address="localhost", port=50040)
    await drone.connect()
    drone1 = System(mavsdk_server_address="localhost", port=50041)
    await drone1.connect()

    # This waits till a mavlink based drone is connected
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone with UUID: {state.uuid}")
            break

    # Checking if Global Position Estimate is ok
    async for global_lock in drone.telemetry.health():
        if global_lock.is_global_position_ok:
            print("-- Global position state is good enough for flying.")
            break

    print("-- Arming")
    await drone.action.arm()

    conf = Config(default_height, follow_distance, direction, responsiveness)
    await drone1.follow_me.set_config(conf)

    print("-- Taking off")
    await drone.action.takeoff()

    await asyncio.sleep(1)
    flying_alt = 20.0 #To fly drone 20m above the ground plane
    #goto_location() takes Absolute MSL altitude 
    await drone.action.goto_location(48, 8, flying_alt, 0)
    await asyncio.sleep(5)

    # This waits till a mavlink based drone is connected
    async for state in drone1.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone 2 with UUID: {state.uuid}")
            break

    # Checking if Global Position Estimate is ok
    async for global_lock in drone1.telemetry.health():
        if global_lock.is_global_position_ok:
            print("-- Global position 2 state is good enough for flying.")
            break

    print("-- Arming1")
    await drone1.action.arm()
    print("-- Taking off 1")
    await drone1.action.takeoff()
    await asyncio.sleep(5)
    await drone.follow_me.start()
    await asyncio.sleep(8)

    for x in range(5):
        print("lat long location...")
        async for position in drone.telemetry.position():
            lat = (position.latitude_deg)
            long = (position.longitude_deg)
            print(lat)
            print(long)
            break
        target = TargetLocation(lat, long, 0, 0, 0, 0)
        print("-- Following Target")
        print("Target Location: ", lat, long)
        await drone1.follow_me.set_target_location(target)
        await asyncio.sleep(55)

if __name__ == "__main__":
    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())
`

JonasVautherin, 2020-11-19:
When you run `./mavsdk_server`, one of the first output lines tells you the version