mavlink/MAVSDK #1272: Mission upload progress functionality

divyanshu1234, 2020-12-01:
I'm using MAVSDK-Java to build a ground control app on Android. Currently, while uploading the `MissionPlan` there is no way to send feedback to the users which tells how much of the upload task has been completed. 

Would it be possible to add a feature to the SDK that shows the upload progress. For example `x` out of `n` `MissionItems` have been uploaded? Especially while uploading large missions or using slow telemetry modules, the user will be able to get an estimate of how much time is required for the task to be completed.

julianoes, 2020-12-01:
@JonasVautherin we could do this by using the finite stream like calibration, right?
 
https://github.com/mavlink/MAVSDK-Proto/blob/88e2fd133d868d3557541d5047549f5dea457685/protos/calibration/calibration.proto#L13-L16

divyanshu1234, 2021-05-26:
Just checking in. Is a feature like this in development?

julianoes, 2021-05-26:
It is not, not that I know of.

JonasVautherin, 2021-05-27:
No, but that would be a nice addition :blush:. Mission upload/download can be very slow sometimes :see_no_evil:

divyanshu1234, 2021-06-09:
Yes, the upload speeds were quite slow, and therefore I started logging and comparing the mission upload messages that QGC sends to the drone with what MAVSDK sends (for simple navigation missions). I noticed that MAVSDK was generating too many `MAV_CMD_DO_CHANGE_SPEED` messages, basically one for each `MAV_CMD_NAV_WAYPOINT`. In comparison, QGC sends the message only once (when you exactly specify when to change the speed), and the rest of the mission is carried out with the same speed.

Over [here](https://github.com/mavlink/MAVSDK/blob/e93f10839a1511fdcfeba59eed381c8bfb6a952f/src/plugins/mission/mission_impl.cpp#L324) in `mission_impl.cpp`, the command is being generated for every mission item which has non `NaN` speed. So basically need to set the speed only in the first mission item, and `NaN` in the remaining.

As a user, I never noticed this. Also, the examples in MAVSDK-Java and MAVSDK-Python never mention this.

It would be nice if this is documented (my bad if I wasn't able to find it). And it would be even nicer if some optimization is implemented in MAVSDK to avoid repeated commands from being generated. Just by doing this for speed, the effective mission size can be reduced to half (for large missions).

I noticed that this is the case with some other parameters too.

JonasVautherin, 2021-06-09:
I think that this is a really good point, it would certainly be a worthwhile optimization!

However, I don't really get why this should be documented. Can't it stay as an implementation detail?

divyanshu1234, 2021-06-09:
Okay. Not exactly documented. But if you want, I can alter the Java client and Python mission examples (with appropriate comments) to highlight this point.

julianoes, 2021-06-10:
We could indeed optimize that. The problem with optimizing it away is the mapping back from mavlink items to MissionItem. At that point you don't know if you had specifically set it or if it should be NAN.

paulholmstedt, 2021-12-21:
Any news on this request? It would indeed be very nice to have a callback similar to subscribe_mission_progress during upload and download of mission so that you know that for example waypoint 10 of 150 has been uploaded (or downloaded). Right now I can't figure out how to indicate the upload or download progress to the operator.

julianoes, 2021-12-21:
@paulholmstedt no news but it just got requested independently from you today, so there might be some movement in the next days.

julianoes, 2022-01-14:
Implemented in #1667.