mavlink/MAVSDK #1280: Pass from Mission mode to Offboard mode

stokekld, 2020-12-04:
Hello everyone.

I would like to know the right way to pass from mission mode to offboard mode and vice versa. I would like to move the drone in the middle of the mission.

Is there a possible way??

julianoes, 2020-12-05:
To engage you offboard mode, you can use `offboard->start()` (https://github.com/mavlink/MAVSDK/blob/d6bbe6acb0fb5f9d3b16c5ef7abd09b43d35aa61/src/plugins/offboard/include/plugins/offboard/offboard.h#L294-L301), to go back to mission mode, just do `mission->start()` (https://github.com/mavlink/MAVSDK/blob/d6bbe6acb0fb5f9d3b16c5ef7abd09b43d35aa61/src/plugins/mission/include/plugins/mission/mission.h#L271-L280).

stokekld, 2020-12-09:
Hey, thank you. I tried what you said and it worked correctly.

julianoes, 2020-12-09:
Awesome, thanks for reporting back.

closed 2020-12-09 by julianoes